robin-shaun/XTDrone #1: Could not get initial transform from base to laser frame, "iris_0/laser"

saengphet, 2020-05-18:
Hello, I tried to run a tutorial, "2D Laser SLAM Simulation" with PX4 firmware v1.11.0-beta1. "roslaunch px4 indoor3.launch" has been executed followed by "roslaunch laser_scan_matcher matcher.launch" in another terminal. I got this error message.

[ WARN] [1589863460.597277470, 2451.336000000]: Skipping scan
[ WARN] [1589863461.607689665, 2452.344000000]: Could not get initial transform from base to laser frame, "iris_0/laser" passed to lookupTransform argument source_frame does not exist.

Nothing appeared in rviz (no expected result as a picture in the tutorial). LaserScan shown status error as "For frame [iris_0/laser]: Frame [iris_0/laser] does not exist" and TF shown warning as "For frame [iris_0/laser]: Frame [iris_0/laser] does not exist"

saengphet, 2020-05-18:
Hello, I tried to run a tutorial, "2D Laser SLAM Simulation" with PX4 firmware v1.11.0-beta1. "roslaunch px4 indoor3.launch" has been executed followed by "roslaunch laser_scan_matcher matcher.launch" in another terminal. I got this error message.

[ WARN] [1589863460.597277470, 2451.336000000]: Skipping scan
[ WARN] [1589863461.607689665, 2452.344000000]: Could not get initial transform from base to laser frame, "iris_0/laser" passed to lookupTransform argument source_frame does not exist.

Nothing appeared in rviz (no expected result as a picture in the tutorial). LaserScan shown status error as "For frame [iris_0/laser]: Frame [iris_0/laser] does not exist" and TF shown warning as "For frame [iris_0/laser]: Frame [iris_0/laser] does not exist".

I appreciated your help !!

robin-shaun, 2020-05-18:
hello, because I have just upgraded the version of px4 and done a lot of modification for XTDrone. So some demos are still waiting to be modified. I will complete the modification in several days.

saengphet, 2020-05-18:
> Hello, I have just upgraded the version of px4 and done a lot of modification for XTDrone, but some demos are still waiting to be modified. I will complete the modification in several days.

Thank you so much for your very fast response. I am looking forward to testing a modified version. I am going to test other tutorials.

robin-shaun, 2020-05-28:
Hello, the manual is OK now. You can test it.

saengphet, 2020-05-30:
> Hello, the manual is OK now. You can test it.

Hello! Thank you for your update. Unfortunately, I have already tested the newest and still got errors as follows

Firstly, I run "roslaunch px4 indoor3.launch", after that I got many warning messages as
**"Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element."** However, the gazebo simulation looks fine, nothing different from the tutorial

[image]

Secondly, I run "roslaunch laser_scan_matcher matcher.launch" in another terminal, I got error messages
**"[ WARN] [1590852379.820397259, 179.096000000]: Could not get initial transform from base to laser frame, "iris_0/laser" passed to lookupTransform argument source_frame does not exist.
"**

![Screenshot at 2020-05-30 22:35:56](https://user-images.githubusercontent.com/12409718/83332766-2fa8ba80-a2c7-11ea-983b-6a1c74d36f5e.png

Next, check rqt tf tree and rqt graph, I got something different from the tutorial.

[image]

[image]

robin-shaun, 2020-05-30:
hello, it's a bug and I have solved it.          
`<frameName>laser_2d</frameName>`  in XTDrone/sitl_config/models/hokuyo_lidar/model.sdf,
previously, it's laser.

saengphet, 2020-05-31:
> frameName

Wonderful, Solved !!

saengphet, 2020-05-31:
> hello, it's a bug and I have solved it.
> `<frameName>laser_2d</frameName>` in XTDrone/sitl_config/models/hokuyo_lidar/model.sdf,
> previously, it's laser.

Hello, iris_0/laser_2d is OK for now, next is "No transform from [map] to frame [odom]" and "No transform from [map_ned] to frame [odom]"

So that, I could not run "roslaunch laser_scan_matcher matcher_gmapping.launch"

[image]

robin-shaun, 2020-05-31:
odom and map is connected by gmapping， not by laser scan matcher

saengphet, 2020-05-31:
Hi, Now it works fine after installed "sudo apt-get install ros-kinetic-slam-gmapping"

[image]

robin-shaun, 2020-05-31:
ok

closed 2020-05-31 by robin-shaun

zhulei991113, 2020-05-31:
> 嗨，现在，在安装“ sudo apt-get install ros-kinetic-slam-gmapping”后，它可以正常工作。我的错。谢谢 ！
> 
> [image]

hello！I am very happy seeing your comments and result.i have a same print
however,i want to ask you for help
class TF in rviz ,have many warning......though map and others  are ok
   additionally,after output "python laser_transfer.py iris 0 2d",the px4 can print :
INFO  [ecl/EKF] 19804000: reset position to ev position
INFO  [ecl/EKF] 19804000: commencing external vision position fusion
INFO  [ecl/EKF] 19804000: commencing external vision yaw fusion
?

saengphet, 2020-05-31:
@zhulei991113 Hi, after ran the script "python laser_transfer.py iris 0 2d" then the external position from ROS will be sent to px4 state estimation (EKF2). The INFO is telling us that PX4 is received the external position (using vision pipeline). Some of TF there is no link between each other (such as base_link_frd and map). However, it is fine because the important frames are linked together.