robin-shaun/XTDrone #4: Imu values from AirSim

plushpluto, 2020-06-21:
I am trying to replicate this work using Airsim.
I didn't know how to take the imu values from airsim. So, can I use imu from MAVROS instead of Gazebo?

#imu_topic: "/iris_0/imu_gazebo"
imu_topic: "/mavros/imu/data"

robin-shaun, 2020-06-21:
sure, but if using vio, you need to modify vio configuration for imu

plushpluto, 2020-06-21:
Oh!! Can you pls guide me?

This is what i tried:: :)

imu_topic: "/mavros/imu/data"
estimate_extrinsic: 1

QiuMingZS, 2020-06-21:
emm, did you mean that you need imu data from AirSim? Or imu data from Gazebo through MAVROS?

QiuMingZS, 2020-06-21:
If you just want to get imu data through AirSim, you may refer to this code:
[hello_drone.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py)
There are some lines in the code above as follows about imu:
```python
import setup_path 
import airsim
client = airsim.MultirotorClient()
imu_data = client.getImuData()
```

QiuMingZS, 2020-06-21:
If you need the imu data from MAVROS, you may have to get the IMU data using AirSim's API and do some data processing and transfer them using MAVROS.

QiuMingZS, 2020-06-21:
Besides, you also raise an issue here:
[AirSim](https://github.com/microsoft/AirSim)

plushpluto, 2020-06-21:
I would like to try both. At the moment, ROS/Gazebo seems to work. For some reason, Airsim is not completely trustable yet (but looks good though). Anyways, thanks. As you said, I will try and get the IMU from Airsim and transfer them using MAVROS.

QiuMingZS, 2020-06-25:
After you successfully complete, welcome to share your experience here with everyone!

plushpluto, 2020-06-25:
Sure. I have already raised an issue with Airsim group and working with them. Will let you know how it goes. ;)

plushpluto, 2020-07-25:
Multiple people have raised multiple issues regarding this, which is causing some important changes to be made. They are making some changes in their end on the ROS repo. That's why the delay along with some delays due to my paper submission. At the moment, I am following the recent repos https://github.com/MichaelHannalla/airsim_ros_bridge and https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_ros_pkgs for updates. Please look out for changes in https://github.com/microsoft/AirSim/pull/2743 https://github.com/microsoft/AirSim/issues/2369 https://github.com/microsoft/AirSim/issues/2396. Meanwhile, I'll try and keep things updated as well.

closed 2020-10-18 by robin-shaun