robin-shaun/XTDrone #5: Comments and code description.

eng-86, 2020-07-25:
Dear, first of all it is a great job was done here. I read your papers, it is very helpful and I hope you continue developing this work. I have only one suggestion in order to share, contribute, and discuss: I suggest adding comments for your code and some discussion, in addition a block diagram for different parts of code and the communication (including ros topics etc...). This should help many researchers and may be used by different universities. Thank you again.

robin-shaun, 2020-07-25:
Thank you for your approval！
Yes, we are keeping developing the platform and hope you to follow our job. And we will also add comments and 
improve the manual [https://www.yuque.com/xtdrone/manual_en](https://www.yuque.com/xtdrone/manual_en)

eng-86, 2020-07-31:
Dear, I read the manual in the following link: https://www.yuque.com/xtdrone/manual_en , almost all. It is very useful and professional and really a great job done there. I am a master degree robotic engineer planning for Phd program in multi robot cooperation. My previous work was on drones (fault tolerance and path planning) implementing PX4 in real word and I have many publications. The tutorial in the link above is very useful as a user guide, I hope it can be upgraded to be more useful for developers and researchers in order to apply their research work and algorithms. This may boost your work by sharing ideas and linking cite tour work and mention the simulator in their publications.
Thank you for appreciating my ideas, and I hope more success and progress. I wish sharing me your technical email in case of any inquiry.
Please I need more information about the application of the formation algorithm which is linked to the following publication https://arxiv.org/abs/2005.01125 (more discussion how it was implemented and programmed).
Thanks.

robin-shaun, 2020-10-18:
You can refer to [https://www.yuque.com/xtdrone/manual_en/uav_formation](https://www.yuque.com/xtdrone/manual_en/uav_formation)

closed 2020-12-10 by robin-shaun