robin-shaun/XTDrone #7: Why a coordinate frame transform is performed in the function construct_target in file multirotor_communication.py?

Sweetyshadow, 2020-10-07:
In the file, line 119 to line 129:
```python           
            target_raw_pose.position.x = -y
            target_raw_pose.position.y = x
            target_raw_pose.position.z = z

            target_raw_pose.velocity.x = -vy
            target_raw_pose.velocity.y = vx
            target_raw_pose.velocity.z = vz
            
            target_raw_pose.acceleration_or_force.x = afx
            target_raw_pose.acceleration_or_force.y = afy
            target_raw_pose.acceleration_or_force.z = afz
```
It's clear that the function do a coordinate frame transform here. But this transform lets MAV do a different behaviours in the Gazebo simulation, it's actually running at RFU coordinate frame. If we don't do the transform here, then the mav in the simulation is running at a right FLU coordinate frame.

robin-shaun, 2020-10-09:
No, it looks strange but it's necessary. Without this transform, it flies at  RFU .

robin-shaun, 2020-10-09:
In fact, I'm not sure why we should do this transform, but it's necessary.

robin-shaun, 2020-10-18:
Oh, I find that the coordinate frame transform of kinetic-mavros is different from melodic-mavros. The bug is removed now. Thanks!

closed 2020-10-18 by robin-shaun