robin-shaun/XTDrone #9: XTDrone execution problems

antonio-sc66, 2020-12-09:
Hi, I've followed the https://www.yuque.com/xtdrone/manual_en to install XTDrone with PX4 in a fresh Ubuntu 18.04 install, with ROS Melodic.

I had to change the _PX4_Firmware_ references to the new _PX4-Autopilot_ default name in the copy steps (https://www.yuque.com/xtdrone/manual_en/basic_config_1.11#2qN28).

I didn't download the models.zip as it requires a Yuque account, but the manual says it should download them when executing.

PX4 runs Ok with gazebo when running make px4_sitl gazebo but doesn't compile when I copy the modified _gazebo_gimbal_controller_plugin.cpp_. It throws the next errors:
`
/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In constructor ‘gazebo::GimbalControllerPlugin::GimbalControllerPlugin()’:
/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:102:9: error: ‘class gazebo::GimbalControllerPlugin’ has no member named ‘lastImuYaw’
   this->lastImuYaw = 0;
         ^~~~~~~~~~

/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘virtual void gazebo::GimbalControllerPlugin::Init()’:
/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:397:87: error: ‘ImuCallback’ is not a member of ‘gazebo::GimbalControllerPlugin’
   imuSub = node->Subscribe("~/" + model->GetName() + "/imu", &GimbalControllerPlugin::ImuCallback, this);
                                                                                       ^~~~~~~~~~~

/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: At global scope:
/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:402:42: error: variable or field ‘ImuCallback’ declared void
 void GimbalControllerPlugin::ImuCallback(ImuPtr& imu_message)
                                          ^~~~~~

/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:402:42: error: ‘ImuPtr’ was not declared in this scope
/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:402:50: error: ‘imu_message’ was not declared in this scope
 void GimbalControllerPlugin::ImuCallback(ImuPtr& imu_message)
                                                  ^~~~~~~~~~~

/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘void gazebo::GimbalControllerPlugin::OnUpdate()’:
/home/foo/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:477:31: error: ‘class gazebo::GimbalControllerPlugin’ has no member named ‘lastImuYaw’
     this->yawCommand += this->lastImuYaw;
                               ^~~~~~~~~~

At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
ninja: build stopped: subcommand failed.

FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 

cd /home/foo/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/foo/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j2

ninja: build stopped: subcommand failed.

Makefile:217: recipe for target 'px4_sitl_default' failed

make: *** [px4_sitl_default] Error 1
`

Due to this error I restore the original file and try to execute the custom XTDrone launch files, but I get the following errors.
`
roslaunch px4 indoor1.launch
`

`
...

Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.

Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.

Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.

[ INFO] [1607554328.840287734]: waitForService: Service [/gazebo/set_physics_properties] is now available.

[ INFO] [1607554328.870219292]: Physics dynamic reconfigure ready.

[INFO] [1607554328.885633, 0.000000]: Loading model XML from file /home/foo/PX4-Autopilot/Tools/sitl_gazebo/models/iris_stereo_camera/iris_stereo_camera.sdf

[INFO] [1607554328.897723, 0.000000]: Waiting for service /gazebo/spawn_sdf_model

[INFO] [1607554328.902918, 0.000000]: Calling service /gazebo/spawn_sdf_model

[INFO] [1607554328.915104, 1814.801000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name iris_0

[ERROR] [1607554328.916445, 1814.801000]: Spawn service failed. Exiting.

Warning [parser.cc:950] XML Element[parent], child of element[sensor] not defined in SDF. Ignoring[parent]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.

[vehicle_spawn_foo_4964_5305119535928211108-5] process has died [pid 5020, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /home/foo/PX4-Autopilot/Tools/sitl_gazebo/models/iris_stereo_camera/iris_stereo_camera.sdf -model iris_0 -x 0 -y 7.5 -z 1 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_foo_4964_5305119535928211108 __log:=/home/foo/.ros/log/29d1f0a8-3a71-11eb-90cf-5254002085db/vehicle_spawn_foo_4964_5305119535928211108-5.log].
log file: /home/foo/.ros/log/29d1f0a8-3a71-11eb-90cf-5254002085db/vehicle_spawn_foo_4964_5305119535928211108-5*.log

[ INFO] [1607554329.433231180, 1814.869000000]: Camera Plugin: The 'robotNamespace' param was empty

[ INFO] [1607554329.433478758, 1814.869000000]: Camera Plugin: The 'robotNamespace' param was empty

[ INFO] [1607554329.435768017, 1814.869000000]: Camera Plugin (ns = iris_0)  <tf_prefix_>, set to "iris_0"

[ INFO] [1607554329.436079956, 1814.869000000]: Camera Plugin (ns = iris_0)  <tf_prefix_>, set to "iris_0"

[ WARN] [1607554329.443008672, 1814.869000000]: missing <robotNamespace>, set to default: /iris_0/

[ INFO] [1607554329.443051924, 1814.869000000]: <topicName> set to: /iris_0/imu_gazebo

[ INFO] [1607554329.443063386, 1814.869000000]: <frameName> set to: imu_link_stereo

[ INFO] [1607554329.443088563, 1814.869000000]: <updateRateHZ> set to: 500

[ INFO] [1607554329.443102359, 1814.869000000]: <gaussianNoise> set to: 0

[ INFO] [1607554329.443123179, 1814.869000000]: <xyzOffset> set to: 0 0 0

[ INFO] [1607554329.443149619, 1814.869000000]: <rpyOffset> set to: 0 -0 0

Segmentation fault (core dumped)

[gazebo-3] process has died [pid 4994, exit code 139, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/foo/PX4-Autopilot/Tools/sitl_gazebo/worlds/indoor1.world __name:=gazebo __log:=/home/foo/.ros/log/29d1f0a8-3a71-11eb-90cf-5254002085db/gazebo-3.log].
log file: /home/foo/.ros/log/29d1f0a8-3a71-11eb-90cf-5254002085db/gazebo-3*.log
`

I would like to know if you have any info that can help me solve this errors, thanks!

robin-shaun, 2020-12-10:
Hello, for the first problem, did you `git checkout -b xtdrone/dev v1.11.0-beta1`. For the second problem, did you upgrade gazebo, default version in melodic is 9.0 and cannot support indoor1.world

antonio-sc66, 2020-12-10:
Hi @robin-shaun. The first problem was related to a download problem during the `git submodule update --init --recursive`. I deleted the PX4 submodules and performed the init again and it now compiles.

The gazebo version I was using at first was 9.0, then I upgraded to 9.13 and I had the same problems but with the latest stable 9.16 everything seems to be working fine.

Thanks!

closed 2020-12-10 by antonio-sc66

Zheng-jia-shuo, 2022-10-06:
hello,i want to know how to delete PX4 submodules and which file folder should i execute  git submodule update --init --recursive

robin-shaun, 2022-10-06:
In PX4_Firmware folder

Zheng-jia-shuo, 2022-10-06:
I checkout in PX4 folder.

Zheng-jia-shuo, 2022-10-06:
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In constructor ‘gazebo::GimbalControllerPlugin::GimbalControllerPlugin()’:
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:102:9: error: ‘class gazebo::GimbalControllerPlugin’ has no member named ‘lastImuYaw’
   this->lastImuYaw = 0;
         ^~~~~~~~~~
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘virtual void gazebo::GimbalControllerPlugin::Init()’:
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:397:87: error: ‘ImuCallback’ is not a member of ‘gazebo::GimbalControllerPlugin’
   imuSub = node->Subscribe("~/" + model->GetName() + "/imu", &GimbalControllerPlugin::ImuCallback, this);
                                                                                       ^~~~~~~~~~~
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: At global scope:
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:402:42: error: variable or field ‘ImuCallback’ declared void
 void GimbalControllerPlugin::ImuCallback(ImuPtr& imu_message)
                                          ^~~~~~
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:402:42: error: ‘ImuPtr’ was not declared in this scope
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:402:50: error: ‘imu_message’ was not declared in this scope
 void GimbalControllerPlugin::ImuCallback(ImuPtr& imu_message)
                                                  ^~~~~~~~~~~
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp: In member function ‘void gazebo::GimbalControllerPlugin::OnUpdate()’:
/home/zjs/old_px4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:477:31: error: ‘class gazebo::GimbalControllerPlugin’ has no member named ‘lastImuYaw’
     this->yawCommand += this->lastImuYaw;
                               ^~~~~~~~~~

Zheng-jia-shuo, 2022-10-06:
@antonio-sc66 hello,I compile again and git submodule update --init --recursive untile there doesn't show any information.Then I configure by YUQue's XTDrone.Error occured when i compile gazebo_gimbal_controller_plugin.cpp.
the error is about imu.How should i do?

robin-shaun, 2022-10-06:
You can remove all the PX4-Autopilot and then do according to the yuque document
```
git clone https://github.com/PX4/PX4-Autopilot.git
mv PX4-Autopilot PX4_Firmware
cd PX4_Firmware
git checkout -b xtdrone/dev v1.11.0-beta1
git submodule update --init --recursive
make px4_sitl_default gazebo
```

Zheng-jia-shuo, 2022-10-06:
ok,thanks!!!

Zheng-jia-shuo, 2022-10-06:
@robin-shaun i follow yuque but ....
[0/1] Re-running CMake...
-- PX4 version: v1.11.0-beta1
-- PX4 config file: /home/zjs/old_px4/PX4_Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-255-g47624a0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zjs/old_px4/PX4_Firmware/build/px4_sitl_default
[354/509] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.58") found components: system thread filesystem chrono date_time atomic 
CMake Warning (dev) at /usr/local/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:424 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:30 (include)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time chrono atomic 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Found Boost: /usr/include (found version "1.65.1")  
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Found UUID: TRUE  
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Found UUID: TRUE  
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.7
-- Checking for module 'libzip'
--   Found libzip, version 1.1.2
**CMake Error at CMakeLists.txt:34 (find_program):
  Could not find px4 using the following names:**

-- Configuring incomplete, errors occurred!
See also "/home/zjs/old_px4/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/zjs/old_px4/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[427/509] Linking CXX static library s...t_control/libmodules__fw_att_control.a
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/zjs/old_px4/PX4_Firmware/build/px4_sitl_default/build_gazebo && /usr/local/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/zjs/old_px4/PX4_Firmware/Tools/sitl_gazebo && /usr/local/bin/cmake -E touch /home/zjs/old_px4/PX4_Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[437/509] Building CXX object src/syst...ystemcmds__tests.dir/test_matrix.cpp.o
ninja: build stopped: subcommand failed.
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

i don't know what is this

Zheng-jia-shuo, 2022-10-06:
there doesn't show any useful information