robin-shaun/XTDrone #12: Rviz not receiving camera data

buket7, 2021-05-25:
[image]

file:///home/ros-7/Pictures/Screenshot%20from%202021-05-26%2000-38-57.png

I'm working on the Visual Inertial Odometer (VIO) tutorial. Rviz also doesn't receive / vins_estimator / image_track data. How can I solve the problem?

robin-shaun, 2021-05-26:
Hi, does the iris_stereo_camera publish image and imu topic?

buket7, 2021-05-26:
Hi, thank you for your prompt response.

->If I don't get it wrong, these are the topics that share IMU and camera data:
"/ iris_0 / stereo_camera / left / image_raw"
"/ iris_0 / stereo_camera / right / image_raw"
"/ iris_0 / imu_gazebo"

->When I call " rostopic list" command, there are the three topics above among the active topics. But when I use the command "rostopic echo" it gives the following error:
WARNING: no messages received and simulated time is active.
Is / clock being published?

->Likewise, when I use the "rostopic hz" command, I get the following error:
subscribed to [/iris_0/stereo_camera/right/image_raw]
WARNING: may be using simulated time
no new messages

->I don't get an error on the IMU topic

robin-shaun, 2021-05-26:
Maybe some gazebo plugin is not intalled

buket7, 2021-05-26:
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
.
.
hi, could these warnings I get have anything to do with my problem?

buket7, 2021-05-30:
SUMMARY
========

CLEAR PARAMETERS
 * /iris_0/mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /iris_0/mavros/cmd/use_comp_id_system_control: False
 * /iris_0/mavros/conn/heartbeat_rate: 1.0
 * /iris_0/mavros/conn/system_time_rate: 1.0
 * /iris_0/mavros/conn/timeout: 10.0
 * /iris_0/mavros/conn/timesync_rate: 10.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /iris_0/mavros/distance_sensor/laser_1_sub/id: 3
 * /iris_0/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/laser_1_sub/subscriber: True
 * /iris_0/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /iris_0/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /iris_0/mavros/distance_sensor/lidarlite_pub/id: 1
 * /iris_0/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /iris_0/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /iris_0/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /iris_0/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /iris_0/mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /iris_0/mavros/distance_sensor/sonar_1_sub/id: 2
 * /iris_0/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /iris_0/mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /iris_0/mavros/fake_gps/eph: 2.0
 * /iris_0/mavros/fake_gps/epv: 2.0
 * /iris_0/mavros/fake_gps/fix_type: 3
 * /iris_0/mavros/fake_gps/geo_origin/alt: 408.0
 * /iris_0/mavros/fake_gps/geo_origin/lat: 47.3667
 * /iris_0/mavros/fake_gps/geo_origin/lon: 8.55
 * /iris_0/mavros/fake_gps/gps_rate: 5.0
 * /iris_0/mavros/fake_gps/mocap_transform: True
 * /iris_0/mavros/fake_gps/satellites_visible: 5
 * /iris_0/mavros/fake_gps/tf/child_frame_id: fix
 * /iris_0/mavros/fake_gps/tf/frame_id: map
 * /iris_0/mavros/fake_gps/tf/listen: False
 * /iris_0/mavros/fake_gps/tf/rate_limit: 10.0
 * /iris_0/mavros/fake_gps/tf/send: False
 * /iris_0/mavros/fake_gps/use_mocap: True
 * /iris_0/mavros/fake_gps/use_vision: False
 * /iris_0/mavros/fcu_protocol: v2.0
 * /iris_0/mavros/fcu_url: udp://:24540@127....
 * /iris_0/mavros/gcs_url: 
 * /iris_0/mavros/global_position/child_frame_id: base_link
 * /iris_0/mavros/global_position/frame_id: map
 * /iris_0/mavros/global_position/gps_uere: 1.0
 * /iris_0/mavros/global_position/rot_covariance: 99999.0
 * /iris_0/mavros/global_position/tf/child_frame_id: base_link
 * /iris_0/mavros/global_position/tf/frame_id: map
 * /iris_0/mavros/global_position/tf/global_frame_id: earth
 * /iris_0/mavros/global_position/tf/send: False
 * /iris_0/mavros/global_position/use_relative_alt: True
 * /iris_0/mavros/image/frame_id: px4flow
 * /iris_0/mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /iris_0/mavros/imu/frame_id: base_link
 * /iris_0/mavros/imu/linear_acceleration_stdev: 0.0003
 * /iris_0/mavros/imu/magnetic_stdev: 0.0
[... 474 lines cut ...]
[ INFO] [1622379633.032623163, 1817.181000000]: RP: Using MISSION_ITEM_INT
[ INFO] [1622379633.032666607, 1817.181000000]: WP: Using MISSION_ITEM_INT
[ INFO] [1622379633.032720291, 1817.181000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1622379633.032756404, 1817.181000000]: VER: 1.1: Flight software:     010b0080 (746b3124ab000000)
[ INFO] [1622379633.032796723, 1817.181000000]: VER: 1.1: Middleware software: 010b0080 (746b3124ab000000)
[ INFO] [1622379633.032836317, 1817.181000000]: VER: 1.1: OS software:         050400ff (ec417d7466666801)
[ INFO] [1622379633.032874675, 1817.181000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1622379633.032907359, 1817.181000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1622379633.032948421, 1817.181000000]: VER: 1.1: UID:                 4954414c44494e4f
[ WARN] [1622379633.035243006, 1817.185000000]: CMD: Unexpected command 520, result 0
[ INFO] [1622379642.050406387, 1826.181000000]: HP: requesting home position
[ INFO] [1622379647.066137789, 1831.181000000]: GF: mission received
[ INFO] [1622379647.066378903, 1831.181000000]: WP: item #0* F: C: 22 p: 15 0 0 nan x: 400921410 y: -36924520 z: 30
[ INFO] [1622379647.066526714, 1831.181000000]: WP: item #1  F: C: 16 p: 0 0 0 nan x: 400937160 y: -36927200 z: 30
[ INFO] [1622379647.066656085, 1831.181000000]: WP: item #2  F: C: 16 p: 10 0 0 nan x: 400930510 y: -36999530 z: 20
[ INFO] [1622379647.066777310, 1831.181000000]: WP: item #3  F: C: 16 p: 5 0 0 nan x: 400915600 y: -36997410 z: 10
[ INFO] [1622379647.066932481, 1831.181000000]: WP: item #4  F: C: 21 p: 0 0 0 nan x: 400917380 y: -36958780 z: 0
[ INFO] [1622379647.066977452, 1831.181000000]: WP: mission received
[ INFO] [1622379647.069702371, 1831.185000000]: FCU: Mission download request ignored, already active
[ WARN] [1622379648.069664045, 1832.181000000]: RP: timeout, retries left 2
[ INFO] [1622379648.069824885, 1832.181000000]: RP: mission received
[ INFO] [1622379652.076244671, 1836.181000000]: HP: requesting home position
[ INFO] [1622379662.106185454, 1846.181000000]: HP: requesting home position

-> I'm using version 9.18 of GAZEBO, but I think there is a problem with the sdf format.

robin-shaun, 2021-05-30:
You can ignore warnings like this `Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.`

buket7, 2021-05-30:
So what do you think is the reason for not getting camera data?

robin-shaun, 2021-05-30:
May be you can try ` sudo apt install ros-melodic-gazebo* `

buket7, 2021-05-30:
I tried, but unfortunately, Rviz still does not display the image.

robin-shaun, 2021-05-30:
Is iris_stereo_camera in your indoor1.launch?
'arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_stereo_camera/iris_stereo_camera.sdf"'

buket7, 2021-05-30:
<arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/indoor1.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_stereo_camera/iris_stereo_camera.sdf"/>

->Yes, the inside of the indoor1.launch file is as above.

robin-shaun, 2021-05-31:
Emm, that's strange.

YiChenCityU, 2021-08-18:
> ```
> <arg name="est" default="ekf2"/>
> <arg name="vehicle" default="iris"/>
> <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/indoor1.world"/>
> <arg name="sdf" default="$(find [...]

Did you solved the problem? I cannot get image either.

robin-shaun, 2021-08-19:
@YiChenCityU @buket7 What's your Gazebo version? Gazebo 11 is not supported.

robin-shaun, 2022-11-02:
You should remove default gazebo models
```rm -r stereo_camera/ 3d_lidar/ 3d_gpu_lidar/ hokuyo_lidar/```

mengchaoheng, 2022-11-05:
同样的问题在使用Ubuntu18到[三维激光slam](https://www.yuque.com/xtdrone/manual_cn/3d_laserslam)是发生rviz无法显示点云数据的情况，
[image]
此外，[A-LOAM源码编译运行](https://github.com/HKUST-Aerial-Robotics/A-LOAM)正常。@robin-shaun

mengchaoheng, 2022-11-05:
我完成了[激光slam](https://www.yuque.com/xtdrone/manual_cn/3d_laserslam)前面的全部课程，所以环境应该没问题

mengchaoheng, 2022-11-05:
我执行过
```
cd ~/.gazebo/models/
rm -r stereo_camera/ 3d_lidar/ 3d_gpu_lidar/ hokuyo_lidar/
```
但一直都是提示没有此文件。所以不存在这个冲突，其他地方可能还有冲突？

s-jhanvi, 2023-10-05:
Hello,
I am using Ubuntu - 18.04 with ROS-melodic and Gazebo 9.19
I am facing the same issue.

When trying to run, `sudo apt install ros-melodic-gazebo*`, it says that there are gazebo11 dependencies which cannot be resolved. 
Kindly help if you were able to solve the issue.

TIA

yuetudazhubo, 2024-09-05:
> 我执行过
> 
> ```
> cd ~/.gazebo/models/
> rm -r stereo_camera/ 3d_lidar/ 3d_gpu_lidar/ hokuyo_lidar/
> ```
> 
> 但一直都是提示没有此文件。所以不存在这个冲突，其他地方可能还有冲突？

Hello, I am facing the same issue as you. Were you able to resolve it successfully afterwards?

20246688, 2024-11-25:
大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢。用英语怎么说

20246688, 2024-11-25:
> 同样的问题在使用Ubuntu18到[三维激光slam](https://www.yuque.com/xtdrone/manual_cn/3d_laserslam)是发生rviz无法显示点云数据的情况， [image] 此外，[A-LOAM源码编译运行](https://github.com/HKUST-Aerial-Robotics/A-LOAM)正常。@robin-shaun

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢。用英语怎么说

20246688, 2024-11-25:
> 同样的问题在使用Ubuntu18到[三维激光slam](https://www.yuque.com/xtdrone/manual_cn/3d_laserslam)是发生rviz无法显示点云数据的情况， [image] 此外，[A-LOAM源码编译运行](https://github.com/HKUST-Aerial-Robotics/A-LOAM)正常。@robin-shaun

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢