robin-shaun/XTDrone #14: 多机仿真出错

Sigmund-SJTUer, 2021-09-23:
RLException: while processing /home/sjtu402/PX4_Tools/Firmware/launch/single_vehicle_spawn_xtd.launch:
Invalid <param> tag: Cannot load command parameter [model_description]: no such command [['xmlstarlet', 'ed', '-d', '//plugin[@name="mavlink_interface"]/mavlink_tcp_port', '-s', '//plugin[@name="mavlink_interface"]', '-t', 'elem', '-n', 'mavlink_tcp_port', '-v', '4560', '/home/sjtu402/PX4_Tools/Firmware/Tools/sitl_gazebo/models/solo_stereo_camera/solo_stereo_camera.sdf']]. 

Param xml is <param command="$(arg cmd)" name="model_description"/>
The traceback for the exception was written to the log file

老师，请问这种错误怎么解决

robin-shaun, 2021-09-23:
sudo apt install xmlstarlet

Sigmund-SJTUer, 2021-09-23:
谢谢老师，问题已经解决

closed 2021-09-23 by Sigmund-SJTUer