robin-shaun/XTDrone #18: 你好，我在第一次安装编译PX4时出错，但是这个错误让我不太能理解

Yecgaa1, 2021-12-05:
```

gdutelc@gdutelc:~/PX4_Firmware$ make px4_sitl_default gazebo
-- PX4 version: v1.11.0-beta1
-- PX4 config file: /home/gdutelc/PX4_Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3")
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-255-g47624a0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gdutelc/PX4_Firmware/build/px4_sitl_default
[0/740] git submodule src/drivers/gps/devices
[2/740] git submodule src/lib/ecl
[8/740] git submodule mavlink/include/mavlink/v2.0
[9/740] git submodule Tools/sitl_gazebo
[18/740] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem
CMake Warning (dev) at /opt/cmake-3.22.0-linux-x86_64/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /opt/cmake-3.22.0-linux-x86_64/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
-- Found ccd: /usr/include (found suitable version "2.0", minimum required is "2.0")
-- Found fcl: /usr/include (found suitable version "0.5.0", minimum required is "0.3.2")
-- Found assimp: /usr/include (found version "5.0.0")
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1")
-- Looking for ignition-math6 -- found version 6.9.2
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.9.2
-- Looking for ignition-transport8 -- found version 8.2.1
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3")
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4")
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Found UUID: TRUE
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.9.2
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Looking for ignition-common3 -- found version 3.14.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.14.0
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.9.2
-- Looking for ignition-fuel_tools4 -- found version 4.4.0
-- Searching for dependencies of ignition-fuel_tools4
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.68.0")
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Found JSONCPP: TRUE
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Found YAML: TRUE
-- Checking for module 'libzip'
--   Found libzip, version 1.5.1
-- Found ZIP: TRUE
-- Looking for ignition-common3 -- found version 3.14.0
-- Looking for ignition-math6 -- found version 6.9.2
-- Looking for ignition-msgs5 -- found version 5.8.1
CMake Error at CMakeLists.txt:34 (find_program):
  Could not find px4 using the following names:

-- Configuring incomplete, errors occurred!
See also "/home/gdutelc/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/gdutelc/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[175/740] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/ulog_stream.cpp.o
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure /home/gdutelc/PX4_Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
cd /home/gdutelc/PX4_Firmware/build/px4_sitl_default/build_gazebo && /opt/cmake-3.22.0-linux-x86_64/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/gdutelc/PX4_Firmware/Tools/sitl_gazebo && /opt/cmake-3.22.0-linux-x86_64/bin/cmake -E touch /home/gdutelc/PX4_Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[189/740] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/ulog_stream.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:200：px4_sitl_default] 错误 1
```
这里显示我遇到个cmake文件不完整的问题，但是并没有告诉我少了什么
以下是我的尝试
从github和gitee下载它
尝试过make clean
ubuntu 20.04
我使用的ros如下：
```

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9

```

Yecgaa1, 2021-12-05:
这个ros环境不是我自己安装的，它可能存在一些奇怪的问题。如果实在无法解决，我才考虑重装我的生产环境，毕竟需要顾及到一些问题

robin-shaun, 2021-12-05:
我也没看出哪儿有问题，不行就重装吧，也快

Yecgaa1, 2021-12-05:
我仔细看了看
[](https://github.com/PX4/PX4-Autopilot/issues/15448)
[](https://github.com/PX4/PX4-SITL_gazebo/pull/550)
我意识到什么
我电脑默认装的是3.5.x不满足3.10.2的px4最低编译要求
因此我手动升级我的cmake到3.22.X
然后出现这个问题
我看到在仿真平台分支中，Sitl这个子模块已经停留了很久没有和主分支同步了（2020.3）
主分支在cmake更新到3.18.X时更新了sitl这个模块，时间点事2020.7
这意味着现在库里的sitl或许是不支持cmake 3.22.X的（他们也提到3.17及以前都很正常）

我将在未来几天内尝试验证它（等我先应付一下考试先QAQ）

Yecgaa1, 2021-12-23:
是的我通过安装cmake3.17解决了该问题。

closed 2021-12-23 by Yecgaa1

yhyl2050, 2022-06-30:
我也是这个问题，安装cmake 3.17也会报错，安装3.16成功解决。

dongdongbh, 2022-10-14:
直接把sitl_gazebo这个子模块更新为 main branch就行了，main branch支持新的cmake