robin-shaun/XTDrone #22: 运行make px4_sitl_default gazebo 报错，gazebo版本为9

accountZJU, 2022-03-01:
[0/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
CMake Warning (dev) at /usr/local/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.18/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:30 (include)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
CMake Error at CMakeLists.txt:34 (find_program):
  Could not find px4 using the following names:

-- Configuring incomplete, errors occurred!
See also "/home/asus/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/asus/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[1/6] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/asus/PX4_Firmware/build/px4_sitl_default/build_gazebo && /usr/local/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/asus/PX4_Firmware/Tools/sitl_gazebo && /usr/local/bin/cmake -E touch /home/asus/PX4_Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:200：px4_sitl_default] 错误 1

robin-shaun, 2022-03-01:
应该是cmake版本过高了，参考这个issue #18

accountZJU, 2022-03-01:
是的，cmake换成3.17就可以了。
还想问一下教程里的gazebo_ros_pkgs.git 下载不下来，有其他途径可以下载吗？
cd catkin_ws/src/
git clone https://github.com/ros-simulation/gazebo_ros_pkgs.git
cd gazebo_ros_pkgs/
git checkout -b melodic origin/melodic-devel
cd ..
git clone https://github.com/ros-planning/moveit_msgs.git
git clone https://github.com/wg-perception/object_recognition_msgs.git
git clone https://github.com/OctoMap/octomap_msgs.git
catkin build

robin-shaun, 2022-03-01:
github的问题，或许等会儿就能下载了

closed 2022-03-01 by robin-shaun

accountZJU, 2022-03-05:
老师，我git获取都是失败，不能访问，有没有其他的方法。我单独下载软件包，但是后面的指令不能运行，弄了几天了，实在弄不出来，求老师帮忙解答一下

robin-shaun, 2022-03-05:
网上有很多关于连github的办法，你多试试

accountZJU, 2022-03-08:
现在又出现两个问题
git checkout -b melodic origin/melodic-devel
分支 'melodic' 设置为跟踪来自 'origin' 的远程分支 'melodic-devel'。
切换到一个新分支 'melodic'

~/catkin_ws/src$ catkin build
catkin：未找到命令
请老师帮忙解答一下