robin-shaun/XTDrone #26: gazebo

jkxdl, 2022-03-30:
Resource not found: The following package was not found in <arg default="$(find mavlink_sitl_gazebo)/worlds/indoor1.world" name="world"/>: mavlink_sitl_gazebo
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/j/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ROS path [3]=/home/j/catkin_ws/src/PX4_Firmware
ROS path [4]=/home/j/PX4_Firmware/Tools/sitl_gazebo
The traceback for the exception was written to the log file

此为运行roslaunch px4 indoor1.launch的报错指令，请问是啥原因

robin-shaun, 2022-03-30:
为什么PX4_Firmware在catkin_ws/src下了？检查一下.bashrc的路径写的对不对

jkxdl, 2022-03-30:
bashrc路径我也改了，bashrc没有报错

jkxdl, 2022-03-30:
# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color|*-256color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# colored GCC warnings and errors
#export GCC_COLORS='error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01'

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi
source /opt/ros/melodic/setup.bash
###
source ~/catkin_ws/devel/setup.bash
source ~/catkin_ws/src/PX4_Firmware/Tools/setup_gazebo.bash ~/catkin_ws/src/PX4_Firmware/ ~/catkin_ws/src/PX4_Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/catkin_ws/src/PX4_Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4_Firmware/Tools/sitl_gazebo

##
clear

jkxdl, 2022-03-30:
然后就是昨天我跟您说的那个问题，加了clear之后，就没了
GAZEBO_PLUGIN_PATH :/home/j/catkin_ws/src/PX4_Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/j/catkin_ws/src/PX4_Firmware//Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/j/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/j/catkin_ws/src/PX4_Firmware/build/px4_sitl_default/build_gazebo

jkxdl, 2022-03-30:
确实是路径问题，有一行忘了改了，但现在又出了这样的问题
RLException: while processing /home/j/catkin_ws/src/PX4_Firmware/launch/single_vehicle_spawn_xtd.launch:
Invalid <param> tag: Cannot load command parameter [model_description]: no such command [['xmlstarlet', 'ed', '-d', '//plugin[@name="mavlink_interface"]/mavlink_tcp_port', '-s', '//plugin[@name="mavlink_interface"]', '-t', 'elem', '-n', 'mavlink_tcp_port', '-v', '4560', '/home/j/catkin_ws/src/PX4_Firmware/Tools/sitl_gazebo/models/iris_stereo_camera/iris_stereo_camera.sdf']]. 

Param xml is <param command="$(arg cmd)" name="model_description"/>
The traceback for the exception was written to the log file

jkxdl, 2022-03-30:
已解决

closed 2022-03-30 by jkxdl