robin-shaun/XTDrone #27: gazebo

jkxdl, 2022-03-30:
gazebo的版本是9，运行roslaunch px4 indoor1.launch后，gazebo弹出，但是黑屏，之后意外关闭，终端出现以下信息：
[gazebo_gui-3] process has died [pid 18863, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/j/.ros/log/9080dbae-afe9-11ec-9ce2-08d23e6b72d6/gazebo_gui-3.log].
log file: /home/j/.ros/log/9080dbae-afe9-11ec-9ce2-08d23e6b72d6/gazebo_gui-3*.log

这是啥原因呢

jkxdl, 2022-03-30:
现在是弹出gazebo后又闪退了

jkxdl, 2022-03-30:
[gazebo-2] process has died [pid 26235, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/j/catkin_ws/src/PX4_Firmware/Tools/sitl_gazebo/worlds/indoor1.world __name:=gazebo __log:=/home/j/.ros/log/5d3a7728-b020-11ec-8298-08d23e6b72d6/gazebo-2.log].
log file: /home/j/.ros/log/5d3a7728-b020-11ec-8298-08d23e6b72d6/gazebo-2*.log
Aborted (core dumped)
[gazebo_gui-3] process has died [pid 26240, exit code 134, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/j/.ros/log/5d3a7728-b020-11ec-8298-08d23e6b72d6/gazebo_gui-3.log].
log file: /home/j/.ros/log/5d3a7728-b020-11ec-8298-08d23e6b72d6/gazebo_gui-3*.log

现在是这样的问题

robin-shaun, 2022-03-30:
不要把PX4_Firmware放在catkin_ws下，严格按照教程来

jkxdl, 2022-03-31:
j@j-F117-V:~$ sudo apt-get update
E: 文件 list 第 1 行的记录格式有误 /etc/apt/sources.list.d/gazebo-stable.list (Component)
E: 无法读取源列表。

gazebo安装也要按教程安装吗？我按pdf'出现了这样的结果，是否正常

jkxdl, 2022-03-31:
j@j-F117-V:~$ sudo apt-get remove ros-melodic-gazebo*
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
E: 无法定位软件包 ros-melodic-gazebo*
E: 无法按照 glob ‘ros-melodic-gazebo*’ 找到任何软件包
E: 无法按照正则表达式 ros-melodic-gazebo* 找到任何软件包

按那个pdf安装后，发现有依赖问题，之后想重新卸载，用官方命令安装，在运行sudo apt-get remove ros-melodic-gazebo*时就出现了以上问题，然后我在网上找了很多方法都未解决，更换源也未解决，请问您是否有解决办法

jkxdl, 2022-03-31:
打开：146737；关闭；56855；defer: 0; 冲突；0                        o在限定的时间内没能找到解决方案。加把劲再试试？[Y/n] 
运行sudo apt install ros-melodic-gazebo-*也一直显示以上提示

closed 2022-03-31 by jkxdl