robin-shaun/XTDrone #30: keybords control

jkxdl, 2022-04-12:
[ WARN] [1649764707.350161174, 243.640000000]: CMD: Unexpected command 176, result 1
WARN  [commander] Failsafe enabled: no RC
WARN  [navigator] Global position failure: fixed bank loiter
[ INFO] [1649764709.151677323, 245.440000000]: FCU: ARMED by Arm/Disarm component command
[ INFO] [1649764709.152816390, 245.440000000]: WP: reached #0
[ INFO] [1649764709.187624420, 245.474000000]: WP: reached #0
[ERROR] [1649764709.203981522, 245.492000000]: FCU: Failsafe enabled: no RC
[ INFO] [1649764709.252494950, 245.540000000]: FCU: Failsafe mode activated
[ INFO] [1649764709.296888714, 245.584000000]: WP: reached #0
[ERROR] [1649764709.303688324, 245.592000000]: FCU: Global position failure: fixed bank loiter

这是运行键盘控制飞行器时发生的错误，我z轴的速度都增加到5以上了，gazebo里的固定翼飞行器旋翼转了一会儿就停下来了，没飞起来

closed 2022-04-12 by jkxdl

Rainlv, 2022-10-24:
遇到了相同情况，请问解决了嘛

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢