robin-shaun/XTDrone #31: reset问题

xd1002, 2022-04-18:
请问xtdrone有提供重设无人机的位姿、速度的rostopic或rosservice或gazebo plugin吗？以及有无碰撞检测功能（就是在和环境障碍物或别的智能体碰撞后能接收到True或者False的rostopic或rosservice或gazebo plugin）

petertheprocess, 2022-04-18:
- You can use "/gazebo/set_model_state" provided by gazebo_ros pkg to reset the uavs' states, however it may cause some issues when runing with PX4 SITL. https://www.yuque.com/xtdrone/manual_cn/accelerate_sim
- For the collision detecting issue, you can use the “gazebo_ros_bumper” plugin, which is also provided by gazebo_ros pkg.

xd1002, 2022-04-19:
> * You can use "/gazebo/set_model_state" provided by gazebo_ros pkg to reset the uavs' states, however it may cause some issues when runing with PX4 SITL. [...]

Thanks for your reply. I tried out these two suggestions last night, and the "gazebo_ros_bumper" plugin does really satisfy my need. However, during my running the codes in the tutorial https://www.yuque.com/xtdrone/manual_cn/accelerate_sim, I faced up with a strange problem. I got my desired result via the original control_gazebo_models.py code. But when I tried to command a specific velocity to the drone by modifying 
`pose_msg.pose.position.x = 0; pose_msg.pose.position.y = 0; pose_msg.pose.position.z = 3`
to
`pose_msg.twist.linear.x = 0; pose_msg.twist.linear.y = 0; pose_msg.twist.linear.z = 3`
and ran the code, the base link of the drone(iris_stereo_camera) and its fan blades began to shake, and the fan blades was separated from the base link of the drone just like the picture below: 
[image]
In the meantime, Gazebo did not report any error
[image]
Is it my missing some processes while running the codes in the tutorial? How can I fix it?

petertheprocess, 2022-04-19:
It is not recommended to mix the dynamics process of the physics engine with your self-defined kinematics process. If u want to run the simulation just with your simplified dynamics or kinematics, u can follow this link https://answers.ros.org/question/218744/gazebo-disable-physics-or-enable-kinematics-through-codecommand-line/ to turn off the physics engine iteration.

However, turning off the physics may also influence the "gazebo_ros_bumper" plugin (I am not sure, u can just try.). If it is the case, u should write some code to detect the collision by yourself by judging the relative poses from each other (such as AABB algorithm, and OBB algorithm).

Good luck to u!

closed 2022-04-19 by petertheprocess

20246688, 2024-11-25:
> * 你可以使用 gazebo_ros pkg 提供的 “/gazebo/set_model_state” 来重置无人机的状态，但是在使用 PX4 SITL 运行时可能会导致一些问题。https://www.yuque.com/xtdrone/manual_cn/accelerate_sim
> * 对于碰撞检测问题，您可以使用 gazebo_ros pkg 提供的 “gazebo_ros_bumper” 插件。

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢