robin-shaun/XTDrone #33: 三维运动规划

jkxdl, 2022-05-10:
我在运行"rviz -d ego_rviz.rviz"是rviz没有出现点云图
[image]
这是运行后的rviz
有解决办法吗

winnnnnnd, 2023-03-16:
我也遇到了这样的问题

uestyhox1, 2023-04-09:
同样的问题但我的终端有报错 
roslaunch ego_planner single_uav.launch 
... logging to /home/kd/.ros/log/dbcd4c9a-d68b-11ed-96c8-d4548bbd70f4/roslaunch-kd-Dell-G15-5511-18187.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kd-Dell-G15-5511:44667/

SUMMARY
========

PARAMETERS
 * /iris_0_ego_planner_node/bspline/limit_acc: 2.0
 * /iris_0_ego_planner_node/bspline/limit_ratio: 1.1
 * /iris_0_ego_planner_node/bspline/limit_vel: 1.0
 * /iris_0_ego_planner_node/fsm/emergency_time: 1.0
 * /iris_0_ego_planner_node/fsm/fail_safe: True
 * /iris_0_ego_planner_node/fsm/flight_type: 2
 * /iris_0_ego_planner_node/fsm/planning_horizen_time: 3.0
 * /iris_0_ego_planner_node/fsm/planning_horizon: 7.5
 * /iris_0_ego_planner_node/fsm/realworld_experiment: False
 * /iris_0_ego_planner_node/fsm/thresh_no_replan_meter: 1.0
 * /iris_0_ego_planner_node/fsm/thresh_replan_time: 1.0
 * /iris_0_ego_planner_node/fsm/waypoint0_x: 0.0
 * /iris_0_ego_planner_node/fsm/waypoint0_y: -20.0
 * /iris_0_ego_planner_node/fsm/waypoint0_z: 1.0
 * /iris_0_ego_planner_node/fsm/waypoint1_x: 6.0
 * /iris_0_ego_planner_node/fsm/waypoint1_y: 0.0
 * /iris_0_ego_planner_node/fsm/waypoint1_z: 1.5
 * /iris_0_ego_planner_node/fsm/waypoint2_x: 8.0
 * /iris_0_ego_planner_node/fsm/waypoint2_y: -8.0
 * /iris_0_ego_planner_node/fsm/waypoint2_z: 1.5
 * /iris_0_ego_planner_node/fsm/waypoint3_x: 0.0
 * /iris_0_ego_planner_node/fsm/waypoint3_y: -15.0
 * /iris_0_ego_planner_node/fsm/waypoint3_z: 1.0
 * /iris_0_ego_planner_node/fsm/waypoint4_x: -15.0
 * /iris_0_ego_planner_node/fsm/waypoint4_y: 0.0
 * /iris_0_ego_planner_node/fsm/waypoint4_z: 1.0
 * /iris_0_ego_planner_node/fsm/waypoint_num: 1
 * /iris_0_ego_planner_node/grid_map/cx: 320.5
 * /iris_0_ego_planner_node/grid_map/cy: 240.5
 * /iris_0_ego_planner_node/grid_map/depth_filter_margin: 2
 * /iris_0_ego_planner_node/grid_map/depth_filter_maxdist: 5.0
 * /iris_0_ego_planner_node/grid_map/depth_filter_mindist: 0.5
 * /iris_0_ego_planner_node/grid_map/depth_filter_tolerance: 0.15
 * /iris_0_ego_planner_node/grid_map/frame_id: world
 * /iris_0_ego_planner_node/grid_map/fx: 554.254691191
 * /iris_0_ego_planner_node/grid_map/fy: 554.254691191
 * /iris_0_ego_planner_node/grid_map/ground_height: -1
 * /iris_0_ego_planner_node/grid_map/k_depth_scaling_factor: 1000.0
 * /iris_0_ego_planner_node/grid_map/local_map_margin: 10
 * /iris_0_ego_planner_node/grid_map/local_update_range_x: 5.5
 * /iris_0_ego_planner_node/grid_map/local_update_range_y: 5.5
 * /iris_0_ego_planner_node/grid_map/local_update_range_z: 4.5
 * /iris_0_ego_planner_node/grid_map/map_size_x: 100
 * /iris_0_ego_planner_node/grid_map/map_size_y: 100
 * /iris_0_ego_planner_node/grid_map/map_size_z: 5
 * /iris_0_ego_planner_node/grid_map/max_ray_length: 4.5
 * /iris_0_ego_planner_node/grid_map/min_ray_length: 0.1
 * /iris_0_ego_planner_node/grid_map/obstacles_inflation: 0.2
 * /iris_0_ego_planner_node/grid_map/p_hit: 0.65
 * /iris_0_ego_planner_node/grid_map/p_max: 0.9
 * /iris_0_ego_planner_node/grid_map/p_min: 0.12
 * /iris_0_ego_planner_node/grid_map/p_miss: 0.35
 * /iris_0_ego_planner_node/grid_map/p_occ: 0.8
 * /iris_0_ego_planner_node/grid_map/pose_type: 1
 * /iris_0_ego_planner_node/grid_map/resolution: 0.1
 * /iris_0_ego_planner_node/grid_map/show_occ_time: False
 * /iris_0_ego_planner_node/grid_map/skip_pixel: 2
[... 24 lines cut ...]
 * /rosversion: 1.14.13

NODES
  /
    iris_0_ego_planner_node (ego_planner/ego_planner_node)
    iris_0_map_to_world (tf/static_transform_publisher)
    iris_0_traj_server (ego_planner/traj_server)
    iris_0_world_to_ground_plane (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[iris_0_map_to_world-1]: started with pid [18217]
process[iris_0_world_to_ground_plane-2]: started with pid [18218]
process[iris_0_ego_planner_node-3]: started with pid [18219]
process[iris_0_traj_server-4]: started with pid [18224]
hit: 0.619039
miss: -0.619039
min log: -1.99243
max: 2.19722
thresh log: 1.38629
[ INFO] [1681013379.960634810]: Wait for 1 second.
[FSM]: from INIT to WAIT_TARGET
[iris_0_ego_planner_node-3] process has died [pid 18219, exit code -11, cmd /home/kd/catkin_ws/devel/lib/ego_planner/ego_planner_node ~odom_world:=/vins_estimator/odometry ~planning/bspline:=/xtdrone/iris_0/planning/bspline ~planning/data_display:=/xtdrone/iris_0/planning/data_display ~planning/broadcast_bspline_from_planner:=/broadcast_bspline ~planning/broadcast_bspline_to_planner:=/broadcast_bspline ~grid_map/odom:=/xtdrone/iris_0/vins_estimator/odometry ~grid_map/cloud:=/iris_0/pcl_render_node/points ~grid_map/pose:=/iris_0/camera_pose ~grid_map/depth:=/iris_0/realsense/depth_camera/depth/image_raw __name:=iris_0_ego_planner_node __log:=/home/kd/.ros/log/dbcd4c9a-d68b-11ed-96c8-d4548bbd70f4/iris_0_ego_planner_node-3.log].
log file: /home/kd/.ros/log/dbcd4c9a-d68b-11ed-96c8-d4548bbd70f4/iris_0_ego_planner_node-3*.log
[ WARN] [1681013380.988374408, 2024.461000000]: [Traj server]: ready.

uestyhox1, 2023-04-09:
我解决了，是通过少写下面这一步：
cd ~/XTDrone/motion_planning/3d
python ego_transfer.py iris 0
不晓得原理是为啥

uestyhox1, 2023-05-06:
三维运动规划，无报错但rviz标注出来方向，无人机没反应
[image]

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-25:
> 三维运动规划，无报错但rviz标注出来方向，无人机没反应 [image]

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢