robin-shaun/XTDrone #34: a problem regarding the connection between mavros and sitl

XiaotongYU, 2022-05-31:
I got some problems when I run ‘roslaunch px4 mavros_posix_sitl.launch’. It can be executed with no error and turn up the world and iris in gazebo correctly but cannot connect to the QGC.

terminal message under 'roslaunch px4 mavros_posix_sitl.launch'
`quantumy@quantumy:~/catkin_wss/mavros_offboard_ws$ roslaunch px4 mavros_posix_sitl.launch
... logging to /home/quantumy/.ros/log/bd0c116c-e08f-11ec-b423-a1f798d135f7/roslaunch-quantumy-53056.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://quantumy:38523/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
[... 301 lines cut ...]
[ INFO] [1653966850.435566351]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[ INFO] [1653966850.435576599]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1653966850.501530477]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1653966850.502181697]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1653966850.521957967]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1653966850.522855181]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1653966850.555177, 0.000000]: Loading model XML from file /home/quantumy/PX4-Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[INFO] [1653966850.557046, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1653966851.435895341]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1653966851.448221903, 0.008000000]: Physics dynamic reconfigure ready.
[INFO] [1653966851.460430, 0.000000]: Calling service /gazebo/spawn_sdf_model
[INFO] [1653966851.648268, 0.176000]: Spawn status: SpawnModel: Successfully spawned entity
[vehicle_spawn_quantumy_53056_6905434312799540180-5] process has finished cleanly
log file: /home/quantumy/.ros/log/bd0c116c-e08f-11ec-b423-a1f798d135f7/vehicle_spawn_quantumy_53056_6905434312799540180-5*.log`

I suppose that I configure the mavros correctly by ‘roslaunch mavros px4.launch fcu_url:=“udp://:14540@127.0.0.1:14557”’ and I also have the correct source in the ~/.bashrc
`source /opt/ros/noetic/setup.bash
source ~/catkin_wss/catkin_ws_XTDrone/devel/setup.bash
source ~/PX4-Firmware/Tools/setup_gazebo.bash /home/quantumy/PX4-Firmware /home/quantumy/PX4-Firmware/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Firmware/Tools/sitl_gazebo`

I can also execute ‘make px4_sitl_default gazebo’ correctly with QGC auto-connected.

Could you please give some advice? I really appreciate any help you can provide.

XiaotongYU, 2022-05-31:
terminal message under 'rostopic echo /mavros/state'
`header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
connected: False
armed: False
guided: False
manual_input: False
mode: ''
system_status: 0
---
`

robin-shaun, 2022-05-31:
There are a lot of reasons to cause  /mavros/state connected False and QGC not connected. I suggest you check your operation carefully refering to the yueque documents.

closed 2022-06-02 by XiaotongYU

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢