robin-shaun/XTDrone #37: 多無人機編隊無法使用鍵盤控制

SamLin55587, 2022-07-19:
不好意思請問一下 當我在執行多無人機編隊的仿真時，啟用multirotor_keyboard_control.py這個檔案時，
執行 bash multi_vehicle_communication.sh的終端機會報下列錯誤: 
請問該如何解決 
[ERROR] [1658131676.966918, 29.630000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7f9c2f80a0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 112, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 146, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 195, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

[ERROR] [1658131676.967896, 29.632000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7fa9e09780>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 112, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 146, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 195, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

[ERROR] [1658131676.972427, 29.632000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7f782bda50>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 112, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 146, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 195, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

[ERROR] [1658131676.974050, 29.632000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7fb1b680a0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 112, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 146, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 195, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

[ERROR] [1658131676.970412, 29.632000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7f8db19c30>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 112, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 146, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 195, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

[ERROR] [1658131676.976495, 29.632000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7f917cc9b0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 112, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 146, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 195, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

robin-shaun, 2022-07-19:
没有一直报错的话，不影响使用

robin-shaun, 2022-07-20:
单机控制成功了吗

SamLin55587, 2022-07-20:
> 单机控制成功了吗

都是失敗的 不曉得是不是因為arm64架構的原因

robin-shaun, 2022-07-20:
仿真还是建议用intel的cpu，arm的架构，可能会有些问题。arm适用于真机飞行的机载电脑

SamLin55587, 2022-07-20:
> 仿真还是建议用intel的cpu，arm的架构，可能会有些问题。arm适用于真机飞行的机载电脑

瞭解 感謝回覆

closed 2022-08-14 by robin-shaun

Rupepsi, 2023-04-16:
最后解决了吗？我也遇到这个问题了

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢