robin-shaun/XTDrone #38: ubuntu20.04起飞显示offboard没有激活是什么原因？

zerocandle, 2022-08-14:
WARN  [mc_pos_control] Offboard activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed

zerocandle, 2022-08-14:
照着雀语配置一步一步做的

robin-shaun, 2022-08-14:
检查一下这两个话题
rostopic echo /iris_0/mavros/state
rostopic echo /iris_0/mavros/local_position/pose

zerocandle, 2022-08-17:
> 检查一下这两个话题 rostopic echo /iris_0/mavros/state rostopic echo /iris_0/mavros/local_position/pose

老师，echo出来没问题，现在可以起飞了，但是换到offboard模式的时候还是会报这几个警告
CMD:Unexpected command 176, result 0
WARN [mc_pos_control] Offboard activation failed with error: Activation Failed
WARN [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN [mc_pos_control] Altitude-Ctrl activation failed with error: Activation Failed

robin-shaun, 2022-08-17:
这个报警有一次无所谓

zerocandle, 2022-08-18:
> 这个报警有一次无所谓
改了点东西又不能起飞了

现在键盘控制不能起飞，仿真界面报错：
[ WARN] [1660806194.433162204, 2024.293000000]: CMD: Unexpected command 176, result 0
INFO  [logger] Start file log (type: full)
[ INFO] [1660806195.635422470, 2025.493000000]: FCU: ARMED by Arm/Disarm component command
INFO  [logger] Opened full log file: ./log/2022-08-18/07_03_15.ulg
[ INFO] [1660806195.687911677, 2025.545000000]: FCU: [logger] file:./log/2022-08-18/07_03_15.ulg
[ INFO] [1660806196.371584114, 2026.225000000]: HP: requesting home position
[ INFO] [1660806205.716683689, 2035.513000000]: FCU: DISARMED by Auto disarm initiated
[ INFO] [1660806206.432453924, 2036.225000000]: HP: requesting home position
INFO  [logger] closed logfile, bytes written: 574946

robin-shaun, 2022-08-18:
改啥了

wenbin-hhhhh, 2022-12-08:
你好我也遇到相似的问题
[image]

robin-shaun, 2022-12-16:
@wenbin-hhhhh 你的图片上传失败了，看不到

hao158, 2023-12-27:
我现在还有同样的问题，请问解决了吗？

hao158, 2023-12-27:
[image]

david-666-maker, 2024-01-06:
试试rm ~/.ros/eeprom/parameters*
rm -rf ~/.ros/sitl*

GZH-czy, 2024-06-17:
> 老师，echo出来没问题，现在可以起飞了，但是换到offboard模式的时候还是会报这几个警告 CMD:Unexpected command 176, result 0 WARN [mc_pos_control] Offboard activation failed with error: Activation Failed WARN [mc_pos_control] [...]

怎么起飞的，我和你一摸一样，都没问题报这几个错，但还是起飞不了

20246688, 2024-11-18:
> 怎么起飞的，我和你一摸一样，都没问题报这几个错，但还是起飞不了

大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-25:
> 试试rm ~/.ros/eeprom/parameters* rm -rf ~/.ros/sitl*

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢