robin-shaun/XTDrone #39: 啟用python multirotor_keyboard_control.py iris 9 vel 時 multi_vehicle_communication.sh報錯

SamLin55587, 2022-08-16:
我在進行9台的多無人機編隊時遇到這個問題，且進行鍵盤控制時，只有一台無人機能成功起飛，連接QGC時也只有顯示一台飛機有與地面站連線。請問要如何解決?
[image]
[image]
[image]

robin-shaun, 2022-08-17:
多机的launch文件是根据我们提供的脚本生成的吗

robin-shaun, 2022-08-17:
还有PX4固件的版本是按照教程要求的吗

closed 2022-09-16 by robin-shaun

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢