robin-shaun/XTDrone #41: ubuntu20.04 make px4_sitl gazebo报错

clnch, 2022-09-13:
sitl_gazebo/src/gazebo_gimbal_controller_plugin.cpp:633:11: error: ‘class gazebo::GimbalControllerPlugin’ has no member named ‘yawPub’; did you mean ‘yawPid’?
  633 |     this->yawPub->Publish(m);
      |           ^~~~~~
      |           yawPid
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /home/chc/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/chc/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/chc/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 5
ninja: build stopped: subcommand failed.
make: *** [Makefile:235：px4_sitl] 错误 1

robin-shaun, 2022-09-13:
按照文档要求，git checkout到指定版本

peitianyu, 2023-08-16:
你好, 已经切过来了, 但还是出现这个问题, 是切换到"git checkout -b xtdrone/dev v1.11.0-beta1"这个版本吗?

BobYounger, 2025-01-21:
> 按照文档要求，git checkout到指定版本

按照文档直接下载的安装包，请问还需要git check吗？