robin-shaun/XTDrone #42: （ubuntu18.04）按照“仿真平台基础配置”教程编译PX4固件报错。

zondafeng, 2022-09-14:
环境配置均按照“仿真平台基础配置”这篇文章进行的。已经完成的步骤如下：
1、已安装ROS melodic。
2、已安装gazebo，但这一步还未进行“我们对Gazebo的ROS插件进行了修改，因此需要源码编译”。
3、已安装MAVROS。
进行到第4步“PX4配置”时，下载了您在语雀上提供的可以直接编译的固件，当我切换到“~/PX4_Firmware”目录下，执行“make px4_sitl_default gazebo”命令，报错如下
[image]
尝试了执行一下“make clean”和“make distclean”命令之后重新执行“make px4_sitl_default gazebo”命令，仍无效果。

zondafeng, 2022-09-14:
之前只顾着看后面的“Makefile:198: recipe for target 'px4_sitl_default' failed'"了。应该再往上翻一下，上面会提示
“**CMake Error at CMakeLists.txt:42 (find_package):
By not providing “Findgazebo.cmake” in CMAKE_MODULE_PATH this project has
asked CMake to find a package configuration file provided by “gazebo”, but
CMake did not find one.**”
网上搜了一下，解决办法是执行“sudo apt-get install libgazebo7-dev”，但是这个会提示你没有这个软件包，后来仔细看了下，他这里是“libgazebo7-dev”，而我按照语雀教程安装的是“gazebo9”，所以我把里面的“7”换成了“9”，执行命令“**sudo apt-get install libgazebo9-dev**”就可以了，之后再执行“make px4_sitl_default gazebo”就可以编译了。但这里第一次编译的话，弹出gazebo窗口之后，软件会一直在初始化，建议关了软件，再执行一次“make px4_sitl_default gazebo”，就可以看到小飞机了。

closed 2022-09-16 by robin-shaun

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢