robin-shaun/XTDrone #47: Could i use the project in the ubuntu20.04 noetic?

lijoe123, 2022-11-01:
Thank you for your work in the ROS. 
I had read the md, and it show me that it can work in Kinetic and Melodic. Could i use the project in the noetic? And do i need to fix something in the XTDrone?
Thank you.

robin-shaun, 2022-11-05:
Sure
[image]

lijoe123, 2022-11-06:
Thank you for your answer.
When i run the follow command

```
cd XTDrone
git submodule update --init --recursive
cp sensing/gimbal/gazebo_gimbal_controller_plugin.cpp ~/PX4_Firmware/Tools/sitl_gazebo/src/
cp sitl_config/init.d-posix/rcS ~/PX4_Firmware/ROMFS/px4fmu_common/init.d-posix/
cp sitl_config/worlds/* ~/PX4_Firmware/Tools/sitl_gazebo/worlds/
cp -r sitl_config/models/* ~/PX4_Firmware/Tools/sitl_gazebo/models/ 
cp -r sitl_config/launch/* ~/PX4_Firmware/launch/
cd ~/.gazebo/models/
rm -r stereo_camera/ 3d_lidar/ 3d_gpu_lidar/ hokuyo_lidar/
```
and then run make` px4_sitl_default gazebo ` 
it occur this problem:
[image]
But the px4  is working before. Could you please tell me how to fix the problem?

robin-shaun, 2022-11-06:
Did you git checkout  PX4 firmware

lijoe123, 2022-11-07:
Thank you for your answer .  
I had followed your step and download the px4.zip which you had provided.
The first time I had run `"make px4_sitl_default gazebo"` succeed, but when I run  `roslaunch px4 mavros_posix_sitl.launch` ,
It occur this problem 
`[gazebo-2] process has died [pid 152773, exit code 255, cmd /opt/ros/noetic/`.
And it can launch the gazebo, but it occur two iris in the gazebo.
The second time, I remove the PX4_Fireware, and unzip it again. 
When I run `make px4_sitl_default gazebo`, it error:
[image]
And git checkout:
[image]

lijoe123, 2022-11-08:
Thanks a lot, I had ran `make px4_sitl_default gazebo` succeed.