robin-shaun/XTDrone #48: VIO无法收到双目相机的数据

Rainlv, 2022-11-01:
系统：WSL2-Ubuntu18

VINS-Fusion 官方DEMO已跑通，如图：
[image]

XTDrone键盘控制也没问题...

飞控EKF已按照教程设置为视觉定位：
```
	# GPS used
	#param set EKF2_AID_MASK 1
	# Vision used and GPS denied
	param set EKF2_AID_MASK 24

	# Barometer used for hight measurement
	#param set EKF2_HGT_MODE 0
	# Barometer denied and vision used for hight measurement
	param set EKF2_HGT_MODE 3
```

运行记录如下：
```
❯ roslaunch px4 indoor1.launch

... logging to /home/i/.ros/log/43386a1a-59c6-11ed-a997-00155df5312e/roslaunch-PC-i-5072.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PC-i:46361/

SUMMARY
========

CLEAR PARAMETERS
 * /iris_0/mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /iris_0/mavros/camera/frame_id: base_link
 * /iris_0/mavros/cmd/use_comp_id_system_control: False
 * /iris_0/mavros/conn/heartbeat_rate: 1.0
 * /iris_0/mavros/conn/system_time_rate: 1.0
 * /iris_0/mavros/conn/timeout: 10.0
 * /iris_0/mavros/conn/timesync_rate: 10.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /iris_0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /iris_0/mavros/distance_sensor/laser_1_sub/id: 3
 * /iris_0/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/laser_1_sub/subscriber: True
 * /iris_0/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /iris_0/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /iris_0/mavros/distance_sensor/lidarlite_pub/id: 1
 * /iris_0/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /iris_0/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /iris_0/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /iris_0/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /iris_0/mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /iris_0/mavros/distance_sensor/sonar_1_sub/id: 2
 * /iris_0/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /iris_0/mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /iris_0/mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /iris_0/mavros/fake_gps/eph: 2.0
[... 482 lines cut ...]
/iris_0/mavros/statustext/send
/iris_0/mavros/target_actuator_control
/iris_0/mavros/terrain/report
/iris_0/mavros/time_reference
/iris_0/mavros/timesync_status
/iris_0/mavros/trajectory/desired
/iris_0/mavros/trajectory/generated
/iris_0/mavros/trajectory/path
/iris_0/mavros/tunnel/in
/iris_0/mavros/tunnel/out
/iris_0/mavros/vfr_hud
/iris_0/mavros/vision_pose/pose
/iris_0/mavros/vision_pose/pose_cov
/iris_0/mavros/vision_speed/speed_twist_cov
/iris_0/mavros/wind_estimation
/rosout
/rosout_agg
/tf
/tf_static
```
可以发现并没有发布`/iris_0/stereo_camera/left/image_raw`和`/iris_0/stereo_camera/right/image_raw`话题数据

另外，在启动的gazebo中`Insert`Tab中，也没有发现`iris_stereo_camera`模型
[image]
但是在

Rainlv, 2022-11-01:
相关issue： #12， 当时好像并没有解决

robin-shaun, 2022-11-02:
You should remove default gazebo models
rm -r stereo_camera/ 3d_lidar/ 3d_gpu_lidar/ hokuyo_lidar/

Rainlv, 2022-11-02:
thanks very much, it works!

closed 2022-11-02 by Rainlv