robin-shaun/XTDrone #49: 为何不更新？

mengchaoheng, 2022-11-03:
老师您好，我想在最新的px4上使用本项目，看到项目文档还是旧版本的px4, 想问既然没有时间办法维护，为何不直接推送到px4官方？

robin-shaun, 2022-11-03:
因为对于仿真，现在用的版本的PX4足够了，XTDrone主要是关注上层的算法，PX4在底层的算法更新影响不大。不过我们确实最近遇到了一些问题，准备更新到master，不过需要一段时间。

mengchaoheng, 2022-11-03:
不仅是px4版本，现在ubuntu18有点旧了，20、22逐渐是主流，在已经使用px4开发环境安装脚本部署完开发环境后，想继续用本项目很困难。如果能保持更新的话最好不过。谢谢老师！！！另外本项目如果能尽可能减少对px4的修改的话就更完美 了。现在px4已经不推荐使用mavros，而是使用ros2。未来谁能占住先机谁的影响力就更大，像阿木实验室的项目已经超过本项目，令人深思

robin-shaun, 2022-11-03:
支持ubuntu20.4啊，22.04确实没适配，太新了。ros2我们也有跟进啊，目前是ros1与ros2混合架构，详见https://www.yuque.com/xtdrone/manual_cn/uav_formation_ros2，完全迁移到ros2现在还不现实，各种开源算法还是以ros1为主。至于阿木实验室他们本来就是公司，star更多也是自然。

mengchaoheng, 2022-11-04:
[image]
老师您好，我进行到vins一节了，但最终没有等来:
"这时PX4仿真终端出现如下所示输出，代表视觉信息融合成功"

```Console
INFO  [ecl/EKF] 1213644000: reset position to ev position
INFO  [ecl/EKF] 1213644000: commencing external vision position fusion
INFO  [ecl/EKF] 1213644000: commencing external vision yaw fusion
```
前面编译允许全部没有报错，请问可能是什么原因？

mengchaoheng, 2022-11-04:
@robin-shaun

mengchaoheng, 2022-11-04:
[image]

mengchaoheng, 2022-11-04:
rostopic echo /iris_0/stereo_camera/left/image_raw 有输出

mengchaoheng, 2022-11-04:
没有/vins_estimator/image_track信息

mengchaoheng, 2022-11-04:
问题出在
```
cd catkin_ws
bash scripts/xtdrone_run_vio.sh
```

robin-shaun, 2022-11-05:
这说明vins编译失败了，麻烦另开一个issue

closed 2022-11-05 by robin-shaun