robin-shaun/XTDrone #50: 三维激光SLAM无法加载传感器数据

mengchaoheng, 2022-11-06:
A-LOAM算法已经按照官网跑通，但在[三维激光SLAM](https://www.yuque.com/xtdrone/manual_cn/3d_laserslam)一节，做好全部准备工作后，运行
```
roslaunch aloam_velodyne aloam_velodyne_HDL_32.launch
```
rviz没有传感器数据，只有相机，但错误如下图
[image]

mengchaoheng, 2022-11-06:
@robin-shaun

robin-shaun, 2022-11-06:
贴一下终端的报错信息

mengchaoheng, 2022-11-06:
[roslaunch_px4_outdoor1.txt](https://github.com/robin-shaun/XTDrone/files/9946272/roslaunch_px4_outdoor1.txt)

mengchaoheng, 2022-11-06:
[roslaunch_px4_outdoor1.txt](https://github.com/robin-shaun/XTDrone/files/9946274/roslaunch_px4_outdoor1.txt)

mengchaoheng, 2022-11-06:
[roslaunch_aloam_velodyne_aloam_velodyne_HDL_32.txt](https://github.com/robin-shaun/XTDrone/files/9946279/roslaunch_aloam_velodyne_aloam_velodyne_HDL_32.txt)

mengchaoheng, 2022-11-06:
vim打开上面两个终端输出文件 @robin-shaun

robin-shaun, 2022-11-07:
没有报错，可能是点云数据没有发布，检查一下激光雷达插件是不是编译好了
[image]

mengchaoheng, 2022-11-07:
> 没有报错，可能是点云数据没有发布，检查一下激光雷达插件是不是编译好了
> 
> [image]
> 
> 

我确定是编译好了，但还不清楚为什么没加载上，而且整个电脑没有同名文件了，所以不存在加载了其他sdf文件或者插件的问题。我再探索一下。刚刚把剩余的其他仿真全都跑了，收获良多，感谢你们团队，做了出色的工作，看得出来你们把整个无人系统仿真研究得很透彻，国内需要更多你们这样的团队！

mengchaoheng, 2022-11-08:
在GazeboRosVelodyneLaser.cpp文件打印调试发现，load函数运行了，但Connectcb()没运行，这里
'''
679010
/ Advertise publisher with a custom callback queue
679010
if (topic name != "") {
679010
679010
ros: :AdvertiseOptions ao = ros: :Advertise0ptions: :create<sensor_msgs: :PointC
topic_name
_, 1,
679010
boost::bind(&GazeboRosVelodyneLaser::Connectcb,this),
5790100
boost::bind(&GazeboRosVelodyneLaser::Connectcb,
this),
5679010
ros: : VoidPtr(),
6679010
&laser_queue.
);
6679010
PUb_ = nh_-advertise(ao);
6679010
6679010
'''
是不是有问题？bind两个一样的函数。

mengchaoheng, 2022-11-08:
@robin-shaun

mengchaoheng, 2022-11-08:
[这两行](https://github.com/robin-shaun/XTDrone/blob/cbcd4d78f129185e8023e79f78d46f00d9987c77/sitl_config/gazebo_plugin/velodyne/velodyne_gazebo_plugins/src/GazeboRosVelodyneLaser.cpp#L189)

mengchaoheng, 2022-11-08:
@robin-shaun

robin-shaun, 2022-12-16:
我知道这个问题出在哪儿了，发布的话题名是/iris_0/velodyne_points，aloam里订阅的话题名velodyne_points。之前三维激光雷达插件一直有问题，导致前缀的iris加不上，只能用于单机仿真，后来改好了，不过aloam代码这里忘改了，目前已更新。

closed 2022-12-16 by robin-shaun

mengchaoheng, 2023-03-04:
更新后依然有问题，我使用opencv 4.2，这个应该没影响吧？原工程的测试都可以跑，仿真中rviz就是没数据。@robin-shaun

mengchaoheng, 2023-03-05:
解决了，原来这个outdoor1.launch文件里：

`<group ns="iris_0">`

`<arg name="vehicle" value="iris"/>`

`<arg name="sdf" value="iris_3d_gpu_lidar"/>`

这三个名字这么讲究，为什么搞这么复杂呢？
```Console
 <!-- 3d_gpu_lidar_0 -->
     <group ns="iris_0">
        <!-- MAVROS and vehicle configs -->
            <arg name="ID" value="0"/>
            <arg name="ID_in_group" value="0"/>
            <arg name="fcu_url" default="udp://:24540@localhost:34580"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/single_vehicle_spawn_xtd.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="1"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="iris"/>
            <arg name="sdf" value="iris_3d_gpu_lidar"/>
```

mengchaoheng, 2023-03-05:
而且飞机飞起来后无法悬停，貌似位置数据不对

mengchaoheng, 2023-03-05:
飞机无法悬停
https://user-images.githubusercontent.com/43166007/222934649-ef0a7170-cde6-460f-93ab-48f2e0c59a22.mov

https://user-images.githubusercontent.com/43166007/222934659-6260a113-8f4e-41bc-bff5-8be563586d7e.mov

mengchaoheng, 2023-03-05:
```Console
[ INFO] [1677975897.138123881]: Plugin waypoint loaded
[ INFO] [1677975897.140332687]: Plugin waypoint initialized
[ INFO] [1677975897.140380023]: Plugin wheel_odometry blacklisted
[ INFO] [1677975897.140460735]: Plugin wind_estimation loaded
[ INFO] [1677975897.140716079]: Plugin wind_estimation initialized
[ INFO] [1677975897.140805458]: Built-in SIMD instructions: ARM NEON
[ INFO] [1677975897.140822501]: Built-in MAVLink package version: 2022.12.30
[ INFO] [1677975897.140831876]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[ INFO] [1677975897.140844877]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1677975897.167077042]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1677975897.167906162]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1677975897.180819694]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1677975897.181793737]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1677975897.269753, 0.000000]: Loading model XML from ros parameter model_description
[INFO] [1677975897.273472, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[friction], child of element[collision] not defined in SDF. Ignoring[friction]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[contact], child of element[collision] not defined in SDF. Ignoring[contact]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:950] XML Element[bounce], child of element[collision] not defined in SDF. Ignoring[bounce]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[... cut ...]
[ INFO] [1677976049.767178905, 1763.532000000]: HP: requesting home position
[ INFO] [1677976054.913291547, 1768.532000000]: GF: mission received
[ INFO] [1677976054.913401843, 1768.532000000]: WP: mission received
[ INFO] [1677976054.913466096, 1768.532000000]: RP: mission received
[ INFO] [1677976060.034087197, 1773.536000000]: HP: requesting home position
[ INFO] [1677976070.441403249, 1783.532000000]: HP: requesting home position
[ INFO] [1677976080.884917463, 1793.532000000]: HP: requesting home position

```

mengchaoheng, 2023-03-05:
起飞后无法悬停是什么原因呢？@robin-shaun

20246688, 2024-11-25:
大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

zhaowangji, 2025-07-27:
avia也不行，在固态激光中，/scan话题没有发布者
有成功的案例，但是也有很多也是/scan没有发布者的案例，是不是哪边更新之后就跑不通了……