robin-shaun/XTDrone #51: 二维运动规划bug

mengchaoheng, 2022-11-06:
~/XTDrone/motion_planning/2d/launch/2d_motion_planning.launch文件的行
 ```
  <arg name="motion_planning" default="/home/mch/XTDrone/motion_planning/2d/"/>
```
应改为
```
 <arg name="motion_planning" default="/home/mch/XTDrone/motion_planning/2d"/>
```

closed 2022-11-06 by robin-shaun

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-18:
> > 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢
> 
> 可能和px4的程序/参数有关，还可能和qgc有关，这个不好说。

那怎么弄捏，有什么建议呢。调试得心累。。。

20246688, 2024-11-18:
> > 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢
> 
> 可能和px4的程序/参数有关，还可能和qgc有关，这个不好说。

感觉这个项目是不是bug非常多啊，我想用键盘控制无人机，是不是只有这个项目最方便额

mengchaoheng, 2024-11-18:
> > > 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢
> > 
> > 
> > 可能和px4的程序/参数有关，还可能和qgc有关，这个不好说。
> 
> 感觉这个项目是不是bug非常多啊，我想用键盘控制无人机，是不是只有这个项目最方便额

印象中不难的，基本使用没问题，现在更新了之后就不清楚了，按理说是使用的特定版本的飞控代码，不会有问题，我也忘了当时是怎么使用。我用键盘飞起来控制没问题。

20246688, 2024-11-19:
OK，我再试试

20246688, 2024-11-19:
> > > > 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢
> > > 
> > > 
> > > 可能和px4的程序/参数有关，还可能和qgc有关，这个不好说。
> > 
> > 
> > 感觉这个项目是不是bug非常多啊，我想用键盘控制无人机，是不是只有这个项目最方便额
> 
> 印象中不难的，基本使用没问题，现在更新了之后就不清楚了，按理说是使用的特定版本的飞控代码，不会有问题，我也忘了当时是怎么使用。我用键盘飞起来控制没问题。

还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不

20246688, 2024-11-20:
> > > > 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢
> > > 
> > > 
> > > 可能和px4的程序/参数有关，还可能和qgc有关，这个不好说。
> > 
> > 
> > 感觉这个项目是不是bug非常多啊，我想用键盘控制无人机，是不是只有这个项目最方便额
> 
> 印象中不难的，基本使用没问题，现在更新了之后就不清楚了，按理说是使用的特定版本的飞控代码，不会有问题，我也忘了当时是怎么使用。我用键盘飞起来控制没问题。

大佬。你知道怎么看四翼无人机运行是GPS定位还是视觉定位或者其他定位呐