robin-shaun/XTDrone #54: 二维激光SLAM仿真在Rviz中没有传感器数据

lijoe123, 2022-11-15:
您好，我ros 的版本是noetic。 我完成视觉SLAM的仿真之后进行二维激光SLAM的仿真，但是不能获取传感器的数据，麻烦您看一下：
终端显示：
` [INFO] [1668488569.184281112, 2222.168000000]: lookupTransform base_stabilized to iris_0/laser_2d timed out. Could not transform laser scan into base_frame.`
[image]

谢谢您

lijoe123, 2022-11-15:
我好像解决了，我在indoor3.launch 里面将fcu_url 改成 udp://:14540@localhost:14557，之后，就可以出现传感信息了

但是，我记得在之前的一个问题看到，不建议将这个fcu_url修改，不太清楚为什么？

mengchaoheng, 2022-11-16:
请问你的三维激光slam可以使用吗？我的一直有问题，无法加载传感器数据。详情见[三维激光问题](https://github.com/robin-shaun/XTDrone/issues/50)

@lijoe123

lijoe123, 2022-11-16:
您好，我没有仿真三维的，你可以尝试一下我的方法？改一下**url**?

mengchaoheng, 2022-11-18:
> 您好，我没有仿真三维的，你可以尝试一下我的方法？改一下**url**?

@lijoe123 不是url的问题，已经调试了插件，是激光传感器插件的问题。

lijoe123, 2022-11-18:
> > 您好，我没有仿真三维的，你可以尝试一下我的方法？改一下**url**?
> 
> @lijoe123 不是url的问题，已经调试了插件，是激光传感器插件的问题。

具体怎么实现可以请教一下吗

mengchaoheng, 2022-11-18:
复现问题吗？你去跑一下[三维激光的例程](https://www.yuque.com/xtdrone/manual_cn/3d_laserslam)就知道了。不会只有我有这个问题吧？我目前没有解决这个问题。调试详情可以参考我的另一个[issue](https://github.com/robin-shaun/XTDrone/issues/50)

mengchaoheng, 2022-11-18:
不知道作者为什么要改激光雷达插件，我感觉官方的插件挺好呀

lijoe123, 2022-11-19:
@mengchaoheng 兄弟，我更离谱，就加载不了相机，也出现跟您一样的问题
[image]

mengchaoheng, 2022-11-19:
> @mengchaoheng 兄弟，我更离谱，就加载不了相机，也出现跟您一样的问题
> 
> [image]
> 
> 

@lijoe123 看来不是我的问题，但似乎老师没空管这个问题

lijoe123, 2022-11-19:
兄弟你是 noetic版本吗， 我二维路径规划 Rviz 会闪退，你有遇过这个问题吗
[image]

mengchaoheng, 2022-11-19:
我没有用20跑这个，我用16把项目跑通了就差三维激光

lijoe123, 2022-11-19:
强强强！！！！！，不知道怎么办

mengchaoheng, 2022-11-19:
我用20单独试了三维激光也不行。其他没试

HuaMeiShan, 2023-03-06:
就是用hector来建图的时候，出现了和你给出一模一样的报错，说无法获取到传感器信息，我也把fcu_url 改回官方的值试了还是不行，请问还有什么方法可以解决吗 好难受 卡了很久了

mengchaoheng, 2023-03-06:
不是改那个，是看：
[image]

HuaMeiShan, 2023-03-06:
这个地方我改了的 这是二维路径规划那地方用到的 但是我现在卡在二维激光雷达建图了 输入
roslaunch hector_slam_launch hector_slam_xtdrone.launch之后 他就报你之前的那个无法获取传感器数据的错误，导致rviz都打不开

 
原来/home/robin路径是作者的电脑存放xtdrone的路径，改成你自己的

HuaMeiShan, 2023-03-06:
恳请楼主给个联系方式 请您帮忙调一下这个问题 有偿报答！

mengchaoheng, 2023-03-06:
@HuaMeiShan 报错时需检查路径是否对，地图服务是否开，可以取消args="-d $(arg motion_planning)/rviz/2d_motion_planning.rviz"手动加载config。

mengchaoheng, 2023-03-06:
就三点比较重要： 
1. 2d_motion_planning.launch

<arg name="motion_planning" default="/home/parallels/XTDrone/motion_planning/2d"/>

2. 2d_motion_planning.launch

<node pkg="map_server" name="map_server" type="map_server" args="$(arg motion_planning)/map/indoor3.yaml"/> 

3. 2d_motion_planning.launch
删除args="-d $(arg motion_planning)/rviz/2d_motion_planning.rviz"，打开rviz后手动加载config。也不知道什么原因。反正上述三点之后就可以了。

uestyhox1, 2023-03-28:
> @mengchaoheng 兄弟，我更离谱，就加载不了相机，也出现跟您一样的问题 [image]

我也是相同的问题请问应该如何解决呀

mengchaoheng, 2023-03-28:
仔细看看上面的说明，路径别少斜杠。

Achu170017, 2023-04-18:
好像把PX4_Firmware模型里的hokuyou_lidar的model.sdf中的framename从laser_2d改成base_link能解决

HuaMeiShan, 2023-04-18:
> 好像把PX4_Firmware模型里的hokuyou_lidar的model.sdf中的framename从laser_2d改成base_link能解决

请问能麻烦说一下具体的路径吗 我查了没有找到hokuyou_lidar在哪儿 请给一下完整路径 非常感谢 期待回复！

Achu170017, 2023-04-18:
打错了是hokuyo：Tools/sitl_gazebo/models

HuaMeiShan, 2023-04-18:
> 打错了是hokuyo：Tools/sitl_gazebo/models

我试了一下 还是不行 请问你还尝试过其他解决方法吗

Achu170017, 2023-04-19:
除了上面的方法，没试过别的方法了

ricklessboy, 2023-09-12:
> @mengchaoheng 兄弟，我更离谱，就加载不了相机，也出现跟您一样的问题 [image]

兄弟，请问你这个问题是怎么解决的呢，我这边也是一样的问题。

ricklessboy, 2023-09-12:
> 兄弟你是 noetic版本吗， 我二维路径规划 Rviz 会闪退，你有遇过这个问题吗 [image]
兄弟，请问您当时这个问题解决了吗？

Philemon1, 2024-03-15:
我使用二维激光SLAM建图出现了lookupTransform base_stabilized to laser_2d timed out. Could not transform laser scan into base_frame.不知道怎么改

potato2105, 2025-02-19:
> 我使用二维激光SLAM建图出现了lookupTransform base_stabilized to laser_2d timed out. Could not transform laser scan into base_frame.不知道怎么改

我看上面有位前辈把PX4_Firmware模型里的hokuyou_lidar的model.sdf中的framename从laser_2d改成base_link，这样我试了试我的也是这样报错，现在解决了。