robin-shaun/XTDrone #55: UBuntu 20.04加载相机模型出错

wenbin-hhhhh, 2022-12-12:
ubuntu 20.04配置VIO模型过程中加载相机模型报错
[image]
[image]

robin-shaun, 2022-12-12:
这俩截图没有报错信息，那些warning不用管，检查一下vins编译和运行是否报错

wenbin-hhhhh, 2022-12-12:
> 

编译过程有两个opencv版本冲突的warn.然后按照例程的话运行python vins_transfer.py iris 0
没有出现
INFO  [ecl/EKF] 1213644000: reset position to ev position
INFO  [ecl/EKF] 1213644000: commencing external vision position fusion
INFO  [ecl/EKF] 1213644000: commencing external vision yaw fusion
这些信息

wenbin-hhhhh, 2022-12-12:
谢谢！问题解决了，是vins编译的问题

closed 2022-12-14 by robin-shaun

20246688, 2024-11-25:
> ubuntu 20.04配置VIO模型过程中加载相机模型报错 [image] [image]

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

20246688, 2024-11-25:
> 谢谢！问题解决了，是vins编译的问题

是什么问题呢，vins需要安装吗还要