robin-shaun/XTDrone #57: vins仿真环节正常，可以使无人机正常悬停，ego-planner出错

RobinHan24, 2022-12-22:
执行命令：roslaunch ego_planner single_uav.launch
报错：[iris_0_ego_planner_node-3] process has died [pid 14317, exit code -11, cmd /home/qlb/UAVProject/catkin_ws/devel/lib/ego_planner/ego_planner_node ~odom_world:=/vins_estimator/odometry ~planning/bspline:=/xtdrone/iris_0/planning/bspline ~planning/data_display:=/xtdrone/iris_0/planning/data_display ~planning/broadcast_bspline_from_planner:=/broadcast_bspline ~planning/broadcast_bspline_to_planner:=/broadcast_bspline ~grid_map/odom:=/xtdrone/iris_0/vins_estimator/odometry ~grid_map/cloud:=/iris_0/pcl_render_node/points ~grid_map/pose:=/iris_0/camera_pose ~grid_map/depth:=/iris_0/realsense/depth_camera/depth/image_raw __name:=iris_0_ego_planner_node 
感谢回复~

ywm5, 2023-02-07:
你好请问有人解决了这个问题吗

2500035435, 2023-02-20:
同问，有人解决了这个问题吗

Balabalaforget, 2023-03-02:
我也遇到了这个问题，把报错的终端关闭，多重复几次roslaunch ego_planner single_uav.launch，有时候会成功。希望有人能解决此问题。

winnnnnnd, 2023-03-16:
我也遇到这个问题🙏，有人知道怎么解决吗？

2500035435, 2023-03-16:
我大概知道是什么问题了，应该是内存不够或者溢出之类，可以将single_uav.launch文件里面的地图大小的value改小，
 `<arg` name="map_size_x" value="100"/>
 `<arg` name="map_size_y" value="100"/>
 `<arg` name="map_size_z" value=" 5"/>
改小之后如果还会报错只能单独关掉single_uav.launch再重启，还不行只能继续改小了。
（目前我是这样做的，有更好方法可以指出）

uestyhox1, 2023-04-09:
我是没有执行——转换相机位姿的坐标系方向这一步：
cd ~/XTDrone/motion_planning/3d
python ego_transfer.py iris 0
就成功了，但是我并不知道为什么少执行这一步相机位姿会成功，求佬解答

jiuhenlipua, 2023-06-03:
同问，请问有人解决了吗

Howtyee, 2023-06-05:
4台iris做集群运动规划时，roslaunch ego_planner multi_uav.launch出现如上报错。通过楼上修改
<arg name="map_size_x" value="20"/>
<arg name="map_size_y" value="20"/>
<arg name="map_size_z" value=" 5"/>
后，运行一秒后再次崩溃出现报错。vins正常。
望解答
======================
ubuntu18.04
px4 11.1
16GB RAM
RTX4070(Laptop)
i9-13900hk
CUDA12.0.1
CUDNN8.9

2500035435, 2023-06-12:
感觉跟配置没多大关系，毕竟配置已经够好了，上面问题主要还是在ego-planner的map部分。因为地图大小固定住了，如果vins出现漂移的话，很容易报错，（毕竟一般高度设的数值不大，而vins漂移在z轴变化很大）建议先看看vins漂移的情况。

Howtyee, 2023-06-12:
但单机模拟时把报错的终端关闭，多重复几次roslaunch ego_planner single_uav.launch，有时候确实会成功，也不会有异常。多机时最多只有一个能够成功。报错表现确实像什么东西溢出了。

2500035435, 2023-06-13:
你可以用rostopic检查一下vins的imu_ropagate话题（不知道仿真有没有这个话题），看它xyz漂移情况如何。

ajmddzp, 2023-08-24:
> 

您好，请问您集群运动规划成功了吗，我跑的时候ego_swarm一直在报no odom，请问有遇到相同情况吗

ricklessboy, 2023-10-05:
> 我是没有执行——转换相机位姿的坐标系方向这一步： cd ~/XTDrone/motion_planning/3d python ego_transfer.py iris 0 就成功了，但是我并不知道为什么少执行这一步相机位姿会成功，求佬解答

我也是这个问题，请问您解决了吗

Krosa223, 2025-04-27:
[https://github.com/ZJU-FAST-Lab/Fast-Drone-250/issues/50#issue-1645335280](agx xavier运行ego_planner出现问题 #50
)
开发者在链接问题中回答这种类型报错大概率是里程计漂移造成越界，建议校验里程计精度。
然后还有一种情况造成启动失败就是mavros/vision_odom/odom 话题未发布，我遇到的情况是使用第三方坐标转换脚本时没注意到它把该话题发布为 “odometry”，于是mavros/vision_odom/odom 话题根本没发东西......同样会报这样的错。

Krosa223, 2025-04-27:
https://github.com/https://github.com/ZJU-FAST-Lab/Fast-Drone-250/issues/50#issue-1645335280

开发者在上述链接问题中回答这种类型报错大概率是里程计漂移造成越界，建议校验里程计精度。
然后还有一种情况造成启动失败就是mavros/vision_odom/odom 话题未发布，我遇到的情况是使用第三方坐标转换脚本时没注意到它把该话题发布为 “odometry”，于是mavros/vision_odom/odom 话题根本没发东西......同样会报这样的错。

Krosa223, 2025-04-27:
https://github.com/ZJU-FAST-Lab/Fast-Drone-250/issues/50#issue-1645335280

开发者在上述链接问题中回答这种类型报错大概率是里程计漂移造成越界，建议校验里程计精度。
然后还有一种情况造成启动失败就是mavros/vision_odom/odom 话题未发布，我遇到的情况是使用XTDrone坐标转换脚本时没注意到它把该话题发布为 “odometry”，于是mavros/vision_odom/odom 话题根本没发东西......同样会报这样的错。