robin-shaun/XTDrone #61: 键盘控制反应迟钝

littlebudin, 2023-02-13:
配置好平台后用键盘控制无人机飞行反应很迟钝或者没有反应

robin-shaun, 2023-02-13:
用的px4是1.13还是1.11呢

littlebudin, 2023-02-15:
1.11,问题解决了是我的控制流程不对，飞机飞的太高了，又遇到一个新的问题，运行roslaunch px4 robocup.launch 打开的gazebo仿真的fps在10以下，但是我用的cpu是i9,显卡是英伟达的3090，硬件配置应该没问题

robin-shaun, 2023-02-15:
显卡驱动没装对

ku-zhen, 2023-02-16:
我也有类似的问题，在室内控制反应迟钝，室外会起飞后自动降落（控制台提示：climb to 11 m (10 m above destination)）。但我无法及时悬停，请问可以指导我下么？我的起飞流程：
1. 键盘 i 加速到 0.5
2. 键盘 b 切换 offboard
3. 键盘 t 解锁无人机
4. 键盘 v 起飞 
5. 键盘 s 悬停（貌似此操作并不工作）

robin-shaun, 2023-02-16:
室内3m有透明屋顶，撞屋顶了

ku-zhen, 2023-02-16:
我在室外也是同样的问题，但是我使用键盘停不下来，不知道是不是操作流程不对

[image]

ku-zhen, 2023-02-16:
室外（outdoor4）表现为缓慢起飞，然后中途突然加速上冲，控制台报下面的错误：

> 我在室外也是同样的问题，但是我使用键盘停不下来，不知道是不是操作流程不对
> 
> [image]

robin-shaun, 2023-02-16:
你起飞模式不对啊，别用takeoff模式，用offboard模式

robin-shaun, 2023-02-16:
别按v，如果觉得起飞慢，把起飞速度开到1m／s

ku-zhen, 2023-02-16:
@robin-shaun 感谢，已经解决！

uestyhox1, 2023-03-28:
[image]
[image]
无人机模型没办法起飞，我按键顺序是vbt，之前是起飞后不能正常控制，现在甚至无法起飞

robin-shaun, 2023-03-29:
键盘 i 加速到 0.5
键盘 b 切换 offboard
键盘 t 解锁无人机
键盘 s 悬停

uestyhox1, 2023-03-31:
> 键盘 i 加速到 0.5 键盘 b 切换 offboard 键盘 t 解锁无人机 键盘 s 悬停

感谢你的方案，我是可以正常控制，但在悬停一会儿后大概五分钟，
[image]
它会慢慢转动视角，逐渐失控乱飞，这是为什么呢？
终端有报错：
WARN  [mc_pos_control] PositionControl: invalid setpoints
WARN  [mc_pos_control] Failsafe: blind land

daibaoan, 2024-01-24:
使用坐标真值

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-25:
> 使用坐标真值

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

20246688, 2024-11-25:
> > 键盘 i 加速到 0.5 键盘 b 切换 offboard 键盘 t 解锁无人机 键盘 s 悬停
> 
> 感谢你的方案，我是可以正常控制，但在悬停一会儿后大概五分钟， [image] 它会慢慢转动视角，逐渐失控乱飞，这是为什么呢？终端有报错： WARN [mc_pos_control] PositionControl： invalid setpoints WARN [mc_pos_control] Failsafe： blind land

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

> 室外（outdoor4）表现为缓慢起飞，然后中途突然加速上冲，控制台报下面的错误：
> 
> > 我在室外也是同样的问题，但是我使用键盘停不下来，不知道是不是操作流程不对
> > [image]

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

20246688, 2024-11-25:
> > 键盘 i 加速到 0.5 键盘 b 切换 offboard 键盘 t 解锁无人机 键盘 s 悬停
> 
> 感谢你的方案，我是可以正常控制，但在悬停一会儿后大概五分钟， [image] 它会慢慢转动视角，逐渐失控乱飞，这是为什么呢？终端有报错： WARN [mc_pos_control] PositionControl： invalid setpoints WARN [mc_pos_control] Failsafe： blind land

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

lxl311, 2025-04-26:
你们用键盘控制的时候会不会出现无人机突然就降落了但是桨叶还在转