robin-shaun/XTDrone #62: 无人车运动规划中RVIZ用2d nav goal设置目标点没反应

2500035435, 2023-02-17:
请问一下，看到文档示例效果是选定目标点后会生成导航路径，但我尝试用2d nav goal按钮设定目标点，rviz图里面没反应，几个终端也没反应，看了一下start和goal的话题信息，显示
no messages received and simulated time is active
Is /clock being published？

是我漏掉什么设置了吗？

closed 2023-02-22 by 2500035435

calmelo, 2023-03-24:
> 请问一下，看到文档示例效果是选定目标点后会生成导航路径，但我尝试用2d nav goal按钮设定目标点，rviz图里面没反应，几个终端也没反应，看了一下start和goal的话题信息，显示 no messages received and simulated time is active Is /clock being published？
> 
> 是我漏掉什么设置了吗？

你好，我也遇到了这个问题，请问你解决了吗？可否指教一下怎么解决的？我感觉A*算法那里没有计算出路径

Lec1ercW, 2023-08-24:
您好，我也遇到类似的问题，请问您解决了么？

11hell, 2024-07-18:
请问大家解决了吗

11hell, 2024-07-18:
> 请问一下，看到文档示例效果是选定目标点后会生成导航路径，但我尝试用2d nav goal按钮设定目标点，rviz图里面没反应，几个终端也没反应，看了一下start和goal的话题信息，显示 no messages received and simulated time is active Is /clock being published？
> 
> 是我漏掉什么设置了吗？

大哥，请问你解决了吗？