robin-shaun/XTDrone #65: orbslam2用键盘控制无人机起飞不了

2500035435, 2023-02-22:
步骤是先启动indoor1仿真和orbslam2，再启动通信连接和键盘控制，速度提到0.3以上进入offboard和armed。
现象：无人机桨叶转动但飞不起来，
通信终端有时候会报错：
iris_0: communication initialized
[ERROR] [1677053911.181272, 2257.289000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7fe7cae900a0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 117, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 151, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 200, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

[image]
飞不起来
[image]
状态一直在降落和offboard自动切换

fjc7, 2023-03-30:
怎么解决的呀

20246688, 2024-11-18:
同问

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

Jimmy-L99, 2024-11-18:
当时没解决到，已经毕业了:)

closed 2024-11-25 by Jimmy-L99