robin-shaun/XTDrone #67: roslaunch px4 mavros_posix_sitl.launch会报错，gazebo里面没有飞机模型出现

winnnnnnd, 2023-03-12:
报错内容如下：
[vehicle_spawn_ubt_12287_2646650632614968629-4] process has died [pid 12348, exit code 1, cmd /home/yu/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -sdf -file /home/yu/PX4_Firmware/Tools/sitl_gazebo/models/iris/iris.sdf -model iris -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_ubt_12287_2646650632614968629 __log:=/home/yu/.ros/log/ca3f3084-c0a5-11ed-aba8-1c1bb5c2f1a0/vehicle_spawn_ubt_12287_2646650632614968629-4.log].

closed 2023-03-12 by winnnnnnd

JinXJinLeX, 2023-12-18:
您好！我也遇到了同样的问题，请问您是如何解决的呢？

JinXJinLeX, 2023-12-18:
[vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784-5] process has died [pid 24113, exit code 1, cmd /home/scott/gvins_yolo_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -sdf -file /home/scott/PX4_Firmware/Tools/sitl_gazebo/models/iris/iris.sdf -model iris -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784 __log:=/home/scott/.ros/log/666e6386-9d86-11ee-8504-7c10c9a0f5eb/vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784-5.log].
log file: /home/scott/.ros/log/666e6386-9d86-11ee-8504-7c10c9a0f5eb/vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784-5*.log

JinXJinLeX, 2023-12-18:
[image]
[image]

JinXJinLeX, 2023-12-18:
第一张图是执行了“make px4_sitl_default gazebo”之后出现的，gazebo中出现了飞机。
第二张图是执行了“roslaunch px4 mavros_posix_sitl.launch”之后出现的。

JinXJinLeX, 2023-12-18:
[image]