robin-shaun/XTDrone #70: VINS-Fusion+RTABMap建图部分，启动RTAB建图报错

silkmeloni, 2023-03-26:
运行roslaunch vins rtabmap_vins.launch后报错，看起来是坐标系转换有问题，不知道是哪里配置错了
[image]
VINS-FUSION仿真能运行出来，在其基础上只把仿真地图的launch文件中的sdf模型从stereo改为realsense，其他没有动

uestyhox1, 2023-03-29:
我也有相同的问题
 roslaunch vins rtabmap_vins.launch
... logging to /home/kd/.ros/log/9c9d253e-cdd7-11ed-a245-d4548bbd70f4/roslaunch-kd-Dell-G15-5511-5070.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kd-Dell-G15-5511:41655/

SUMMARY
========

CLEAR PARAMETERS
 * /points_xyzrgb/
 * /rtabmap/rtabmap/

PARAMETERS
 * /points_xyzrgb/approx_sync: True
 * /points_xyzrgb/approx_sync_max_interval: 0.0
 * /points_xyzrgb/decimation: 4.0
 * /points_xyzrgb/voxel_size: 0.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 200
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2

NODES
  /
    iris_0_camera_link_to_depth (tf/static_transform_publisher)
    points_xyzrgb (nodelet/nodelet)
    rviz (rviz/rviz)
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)

ROS_MASTER_URI=http://localhost:11311

process[iris_0_camera_link_to_depth-1]: started with pid [5101]
[... 78 lines cut ...]
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /iris_0/realsense/depth_camera/color/image_raw \
   /iris_0/realsense/depth_camera/depth/image_raw \
   /iris_0/realsense/depth_camera/color/camera_info
[ WARN] [1680056273.117830714, 1941.181000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /iris_0/realsense/depth_camera/color/image_raw \
   /iris_0/realsense/depth_camera/depth/image_raw \
   /iris_0/realsense/depth_camera/color/camera_info
[ WARN] [1680056278.120715940, 1946.181000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /iris_0/realsense/depth_camera/color/image_raw \
   /iris_0/realsense/depth_camera/depth/image_raw \
   /iris_0/realsense/depth_camera/color/camera_info
[ WARN] [1680056283.124468668, 1951.181000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /iris_0/realsense/depth_camera/color/image_raw \
   /iris_0/realsense/depth_camera/depth/image_raw \
   /iris_0/realsense/depth_camera/color/camera_info

[image]

uestyhox1, 2023-03-29:
我也是vio可以但就改了一下launch换成深度相机，就bug了

silkmeloni, 2023-03-29:
我将rtabmap_vins.launch文件中的iris0/depth_camera_base改成了depth_camera_base，问题应该是得到了解决。

silkmeloni, 2023-03-29:
我将rtabmap_vins.launch文件中的iris0/depth_camera_base改成了depth_camera_base，问题应该是得到了解决。
你可以尝试一下，但咱们的报错好像不完全一样，不一定能解决
 

LiJiaXu
***@***.***

uestyhox1, 2023-03-29:
请问这里的修改逻辑是什么呢？是要把iris0全都改掉嘛？
launch文件中：
<launch>
    <node pkg="tf" type="static_transform_publisher" name="iris_0_camera_link_to_depth"
    args="0.0 0.0 0 0.0 0.0 0.0 /camera_link /iris_0/depth_camera_base 40" />
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/iris_0/realsense/depth_camera/depth/image_raw"/>
            <arg name="frame_id"           value="camera_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/vins_estimator/odometry"/>
            <arg name="rgb_topic"          value="/iris_0/realsense/depth_camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/iris_0/realsense/depth_camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>

</launch>

uestyhox1, 2023-03-29:
<launch>
    <node pkg="tf" type="static_transform_publisher" name="iris_0_camera_link_to_depth"
    args="0.0 0.0 0 0.0 0.0 0.0 /camera_link /iris_0/depth_camera_base 40" />
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/iris_0/realsense/depth_camera/depth/image_raw"/>
            <arg name="frame_id"           value="camera_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/vins_estimator/odometry"/>
            <arg name="rgb_topic"          value="/iris_0/realsense/depth_camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/iris_0/realsense/depth_camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>

</launch>

silkmeloni, 2023-03-29:
对，我是把带iris0的全部删掉，可能是tf发布和订阅坐标系话题的名称不对应，我也不太懂，就莫名其妙这样改就改好了。我是第一次做仿真

LiJiaXu
***@***.***

uestyhox1, 2023-03-29:
[image]

uestyhox1, 2023-03-29:
好的我试下，谢谢

uestyhox1, 2023-03-29:
我修改之后还是没有成功，但问题可能还是在这个launch文件中，警告的内容实质上没有变化
[ERROR] [1680060652.867450071, 2162.593000000]: PluginlibFactory: The plugin for class 'octomap_rviz_plugin/ColorOccupancyGrid' failed to load.  Error: According to the loaded plugin descriptions the class octomap_rviz_plugin/ColorOccupancyGrid with base class type rviz::Display does not exist. Declared types are  Submaps moveit_rviz_plugin/MotionPlanning moveit_rviz_plugin/PlanningScene moveit_rviz_plugin/RobotState moveit_rviz_plugin/Trajectory rtabmap_ros/Info rtabmap_ros/MapCloud rtabmap_ros/MapGraph rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu
[ WARN] [1680060657.642496705, 2167.373000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /realsense/depth_camera/color/image_raw \
   /realsense/depth_camera/depth/image_raw \
   /realsense/depth_camera/color/camera_info
[ WARN] [1680060662.645575772, 2172.373000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /realsense/depth_camera/color/image_raw \
   /realsense/depth_camera/depth/image_raw \
   /realsense/depth_camera/color/camera_info
[ WARN] [1680060667.648368084, 2177.373000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /realsense/depth_camera/color/image_raw \
   /realsense/depth_camera/depth/image_raw \
   /realsense/depth_camera/color/camera_info
[ WARN] [1680060672.650552289, 2182.373000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /realsense/depth_camera/color/image_raw \
   /realsense/depth_camera/depth/image_raw \
   /realsense/depth_camera/color/camera_info
[ WARN] [1680060677.653516938, 2187.373000000]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /vins_estimator/odometry \
   /realsense/depth_camera/color/image_raw \
   /realsense/depth_camera/depth/image_raw \
   /realsense/depth_camera/color/camera_info
[ WARN] [16

Randall-W, 2023-11-13:
启动仿真程序的地方不对，直接cd到~/XTDrone/sitl_config/launch，然后roslaunch indoor1.launch (对应的仿真文件)
直接roslaunch px4 indoor1.launch 就一直只有双目的信息没有rgbd信息

20246688, 2024-11-25:
> 我也是vio可以但就改了一下launch换成深度相机，就bug了

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

20246688, 2024-11-25:
> 启动仿真程序的地方不对，直接cd到~/XTDrone/sitl_config/launch，然后roslaunch indoor1.launch （对应的仿真文件） 直接roslaunch px4 indoor1.launch 就一直只有双目的信息没有rgbd信息

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢