robin-shaun/XTDrone #76: Noetic 版本，使用你们提供的px4 11.1版本，无法加载深度摄像头

lijoe123, 2023-04-10:
使用命令roslaunch px4 mavros_posix_sitl.launch，并修改该launch的<sdf>：
` <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_depth_camera/iris_depth_camera.sdf"/>` 但是没有相关的图像话题信息，使用fpv_cam时有图像话题信息，使用irit_triple_depth_camera时，会出现无人机乱飞现象。
[image]
[image]
我想请教一下该怎么给无人机加载深度相机？十分感谢，两个号都给你们star了！！

robin-shaun, 2023-04-11:
用iris_realsense_camera
https://www.yuque.com/xtdrone/manual_cn/sensor_model#6TGWp

lijoe123, 2023-04-11:
十分感谢您的回复，但是我在`/home/music/PX4-Firmware/Tools/sitl_gazebo/models`中没有`iris_realsense_camera`,请问我该怎么添加上去呢？谢谢您。

robin-shaun, 2023-04-11:
仔细看教程
https://www.yuque.com/xtdrone/manual_cn/basic_config#2qN28

lijoe123, 2023-04-11:
十分感谢您的回复，可以调用realsense了，另外，运行过程中出现这个报错：`ERROR [param] Parameter COM_CPU_MAX not found.` 会影响正常使用吗，需要排除这个错误吗？
另外，请问您开发的这个realsense有align depth这个话题吗？我在topic那没发现，还是默认已经调整过了？
谢谢您！

closed 2023-04-12 by lijoe123