robin-shaun/XTDrone #80: roslaunch px4 indoor3.launch时建立通信连接后报错，且飞机不起飞

Achu170017, 2023-04-13:
您好，我这里有问题：执行roslaunch px4 indoor3.launch时，用python multirotor_communication.py iris 0
建立通信连接后报错
[ERROR] [1681371556.786001208, 174.564000000]: PR: Unknown parameter to set: COM_RCL_EXCEPT，但仍可以运行

执行键盘控制无人机程序后，按键后飞机没有反应
rostopic iris_0/mavros/state显示连接为true ，arm为false，按t后仍为false

请问这个问题是怎么回事？

andy-zhuo-02, 2023-04-13:
你好，关于COM_RCL_EXPECT参数的含义可以参见[(https://docs.px4.io/main/en/advanced_config/parameter_reference.html#COM_RCL_EXCEPT)](#COM_RCL_EXCEPT)，简而言之这个参数是关于飞行前安全检查的一个参数。

是只有indoor3.launch文件会出现这个问题，还是其他的launch文件也是这样呢。

又或者，你可以尝试在QGC中直接查看并修改参数（这是一个紧急补救的方法）

Achu170017, 2023-04-13:
感谢您的回答，我这边试过indoor1也是这样的，我现在用QGC把COM_RCL_EXCEPT=4试试

Achu170017, 2023-04-14:
我用QGC连接indoor3和mavros_posix_sitl的无人机，发现都需要加速度计校准，以前都没有，不知道怎么回事

Achu170017, 2023-04-14:
我这边问题已经解决了：
rm ~/.ros/eeprom/parameters*
rm -rf ~/.ros/sitl*
清空仿真参数后重启仿真就好了