robin-shaun/XTDrone #84: 目标检测和跟踪无法加载摄像头

doggystyle-star, 2023-05-05:
老师您好，我在执行：
roslaunch darknet_ros task1.launch和roslaunch px4 outdoor1.launch的时候，并不会出现yolo窗口（如下所示只有飞机前面出现图片），然后px4仿真这个界面出现报错以及进程已经died ，请问是不是typhoon模型的问题
[image]
[image]
[image]

doggystyle-star, 2024-03-03:
> 解决了吗哥们，我也遇到了这个问题
太久没看消息，已经解决请参考https://github.com/doggystyle-star/PX4_yolov5

Shu0T1an, 2024-03-15:
具体咋解决呢呜呜呜

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-25:
> 具体咋解决呢呜呜呜

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

20246688, 2024-11-25:
> 解决了吗哥们，我也遇到了这个问题

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢