robin-shaun/XTDrone #85: 三维运动规划和集群运动规划运行时问题

free26, 2023-05-09:
肖老师，您好，我在做集群运动规划时候发现使用vins-fusion传回点云的时候会出现：无人机在前面空旷的时候会显示出无人机在点云中，但是前面只要有障碍物就只会显示障碍物的点云，不会显示无人机在点云中。导致在进行运动规划时显示无人机在障碍物中而停止规划。如下面两个图所示：
[image]
[image]
在终端有下面这个警告，不知道与这个有没有关系。没有报其他错误。
[image]

free26, 2023-05-09:
集群中的每个无人机都会报这个错
[image]

ajmddzp, 2023-09-18:
您好，我在运行ego_swarm时遇到一些问题：我在尝试roslaunch ego_planner multi_uva.launch后报错no odom，请问您有遇到过吗，是怎么解决的呢？

calmelo, 2023-12-13:
> 肖老师，您好，我在做集群运动规划时候发现使用vins-fusion传回点云的时候会出现：无人机在前面空旷的时候会显示出无人机在点云中，但是前面只要有障碍物就只会显示障碍物的点云，不会显示无人机在点云中。导致在进行运动规划时显示无人机在障碍物中而停止规划。如下面两个图所示： [image] [image] 在终端有下面这个警告，不知道与这个有没有关系。没有报其他错误。 [image] 集群中的每个无人机都会有这个警告 [image]

请问你解决这个问题了吗？我这边一运行就报错the drone is in obstacle.

coca-fanta, 2024-06-17:
> 您好，我在运行ego_swarm时遇到一些问题：我在尝试roslaunch ego_planner multi_uva.launch后报错no odom，请问您有遇到过吗，是怎么解决的呢？

请问您解决 no odom这个问题了吗

zhouleiqiang, 2024-07-17:
> > 您好，我在运行ego_swarm时遇到一些问题：我在尝试roslaunch ego_planner multi_uva.launch后报错no odom，请问您有遇到过吗，是怎么解决的呢？
> 
> 请问您解决 no odom这个问题了吗

你好，您解决no odom这个问题了吗

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢