robin-shaun/XTDrone #86: 采用XTDGroundControl控制多无人机运动

xiaoshengjianan, 2023-05-18:
将xtdrone_qt复制到工作空间进行编译后出现：
 By not providing "Findcontrol.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "control", but
  CMake did not find one.

  Could not find a package configuration file provided by "control" with any
  of the following names:

    controlConfig.cmake
    control-config.cmake

robin-shaun, 2023-05-23:
sudo apt install ros-melodic-control-config

SantuGo, 2023-09-08:
看一下ros版本，ubuntu20.04需要执行sudo apt install ros-noetic-control-config。
如果ubuntu18.04报这个错的话可能是软件源有问题，或者没有apt update

SantuGo, 2023-09-08:
> > sudo apt install ros-melodic-control-config
> 
> E: Unable to locate package ros-melodic-control-config

看一下ros版本，ubuntu20.04需要执行sudo apt install ros-noetic-control-config。
如果ubuntu18.04报这个错的话可能是软件源有问题，或者没有apt update

20246688, 2024-11-25:
请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

20246688, 2024-11-25:
> > > sudo apt install ros-melodic-control-config
> > 
> > 
> > E: Unable to locate package ros-melodic-control-config
> 
> 看一下ros版本，ubuntu20.04需要执行sudo apt install ros-noetic-control-config。 如果ubuntu18.04报这个错的话可能是软件源有问题，或者没有apt update

大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢