robin-shaun/XTDrone #91: 在修改sitl_gazebo中的gazebo_gimbal_controller_plugin.cpp后再次编译时报错

SantuGo, 2023-09-08:
在修改sitl_gazebo中的gazebo_gimbal_controller_plugin.cpp后再次编译时报错，
报错信息有两条，如下：

1. ERROR [param] Parameter COM_CPU_MAX not found.

2. 
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/santu/code/PX4_Firmware//Tools/sitl_gazebo/models/livox_mid40"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/santu/code/PX4_Firmware/Tools/sitl_gazebo/models/livox_mid40"
（此报错信息出现了两次）

此时gazebo已经可以正常运行，请问怎么处理这两条报错信息？

Hiashekiy, 2024-03-02:
我也遇到了，请问楼主解决了吗

marigold36, 2024-03-18:
我在/Tools/sitl_gazebo/models/livox_mid40中自己加入了一个model.config，该报错消失，可是还存在第一个报错

WD1-3, 2024-03-24:
兄弟，你这个model.config文件怎么写的，它这个包不是少了sdf文件吗？

marigold36, 2024-03-26:
> 兄弟，你这个model.config文件怎么写的，它这个包不是少了sdf文件吗？

<?xml version="1.0"?>
<model>
  <name>livox_mid40</name>
  <version>1.0</version>
  <sdf version="1.5">model.sdf</sdf>

  <author>
    <name>Carlos Agüero</name>
    <email>caguero@openrobotics.org</email>
  </author>

  <description>
    A floating buoy for the RobotX competition.  copied from surmak46104
  </description>
</model>

marigold36, 2024-03-26:
> 兄弟，你这个model.config文件怎么写的，它这个包不是少了sdf文件吗？

`<?xml version="1.0"?>
<model>
  <name>livox_mid40</name>
  <version>1.0</version>
  <sdf version="1.5">model.sdf</sdf>

  <author>
    <name>Carlos Agüero</name>
    <email>caguero@openrobotics.org</email>
  </author>

  <description>
    A floating buoy for the RobotX competition.  copied from surmak46104
  </description>
</model>`

stevenakkk, 2024-09-02:
很疑惑，看见有的帖子说第一个错误是不影响的

20246688, 2024-11-25:
请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态