robin-shaun/XTDrone #98: 在Ubuntu20.04+ROS-noetic下学习XTDrone使用文档中“二维激光SLAM（HectorSLAM）”项目时，完全按照教程操作，遇到了tf树和文档不一致的情况

xxhaonan, 2023-12-07:
在运行`roslaunch hector_slam_launch hector_slam_xtdrone.launch`时遇到报错
`lookupTransform base_link to laser_2d timed out. Could not transform laser scan into base_frame`
[image]
查看tf树发现和文档中的tf树不一样
我的：
[image]
文档中的：
[image]
并且在我运行`python multirotor_keyboard_control.py iris 1 vel`时
并不能通过键盘来操纵无人机，望解答

xxhaonan, 2023-12-08:
已解决

closed 2023-12-08 by xxhaonan

hangt777, 2024-07-07:
大佬怎么解决的，我现在也是一样的问题😭😭😭

hangt777, 2024-07-07:
> 已解决
大佬怎么解决的，我现在也遇到了一样的问题😭😭😭