robin-shaun/XTDrone #99: 编译运行后gazebo中没有飞机模型出现

JinXJinLeX, 2023-12-18:
执行“roslaunch px4 mavros_posix_sitl.launch”之后报错
[vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784-5] process has died [pid 24113, exit code 1, cmd /home/scott/gvins_yolo_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -sdf -file /home/scott/PX4_Firmware/Tools/sitl_gazebo/models/iris/iris.sdf -model iris -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 __name:=vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784 __log:=/home/scott/.ros/log/666e6386-9d86-11ee-8504-7c10c9a0f5eb/vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784-5.log].
log file: /home/scott/.ros/log/666e6386-9d86-11ee-8504-7c10c9a0f5eb/vehicle_spawn_scott_System_Product_Name_24008_1332974350459587784-5*.log

JinXJinLeX, 2023-12-18:
执行“make px4_sitl_default gazebo”后的报错和gazebo图片。
[image]

JinXJinLeX, 2023-12-18:
执行“roslaunch px4 mavros_posix_sitl.launch”后的报错和gazebo图片（没有无人机模型）。
[image]

JinXJinLeX, 2023-12-21:
问题已解决，第一个是由于Qt缺少了一个库，使用ldd libxxxx.so查看该库的使用情况，并手动下载缺少的库，使用动态库链接后能成功编译；第二个是由于python环境配置错误，不能使用python3来运行，需要切换到python2环境，否则会出现无人机model编译失败问题，因此无人机模型不出现。

closed 2023-12-21 by JinXJinLeX

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg](https://search.bilibili.com/all?from_source=webcommentline_search&keyword=qcg&seid=3759131396693666756)软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-25:
> 问题已解决，第一个是由于Qt缺少了一个库，使用ldd libxxxx.so查看该库的使用情况，并手动下载缺少的库，使用动态库链接后能成功编译；第二个是由于python环境配置错误，不能使用python3来运行，需要切换到python2环境，否则会出现无人机model编译失败问题，因此无人机模型不出现。

现在好像是因为ros文件权限需要打开