robin-shaun/XTDrone #103: 您好！运行VINS-fusion时，仿真环境飞机图像和imu话题都有数据，但是vins-fusion相关的话题都没有数据

SSZ1, 2024-03-26:
第一个问题：运行indoor1.launch后i，/iris_0/stereo_camera/right/image_raw、/iris_0/stereo_camera/right/image_raw、/iris_0/imu_gazebo三个话题都有数据， 但是vins相关的话题，/vins_estimator/image_track、/vins_estimator/camera_pose都没有数据输出，rviz图像位置不显示，EKF2已修改为视觉导航模式，因为vins无法输出位姿，因此解锁飞机时没有反应。请问这是什么问题？
以下是相关消息数据：
[image]
[image]
[image]

[image]

仿真环境与vins运行窗口为
[image]

`ssz@ssz-G5-5500:~/WorkSpace/Auto-Uav/uav_sim$ bash scripts/xtdrone_run_vio.sh 
[ INFO] [1711424936.583679628]: init begins
config_file: /home/ssz/WorkSpace/Auto-Uav/uav_sim/src/VINS-Fusion/config/xtdrone_sitl/px4_sitl_stereo_imu_config.yaml
USE_IMU: 1
IMU_TOPIC: /iris_0/imu_gazebo
result path /home/ssz/WorkSpace/Auto-Uav/uav_sim/vins_output/vio.csv
[ WARN] [1711424936.616103346]:  fix extrinsic param 
camera number 2
[ INFO] [1711424936.619258060]: Synchronized sensors, fix time offset: 0
[ INFO] [1711424936.619312802]: ROW: 480 COL: 752 
 exitrinsic cam 0
 0  0  1
-1  0  0
 0 -1  0
   0 0.12 -0.3
 exitrinsic cam 1
 0  0  1
-1  0  0
 0 -1  0
   0    0 -0.3
set g       0       0 9.81007
[ INFO] [1711424936.619430516]: reading paramerter of camera /home/ssz/WorkSpace/Auto-Uav/uav_sim/src/VINS-Fusion/config/xtdrone_sitl/cam0_pinhole_p1.yaml
[ INFO] [1711424936.619646728]: reading paramerter of camera /home/ssz/WorkSpace/Auto-Uav/uav_sim/src/VINS-Fusion/config/xtdrone_sitl/cam1_pinhole_p1.yaml
MULTIPLE_THREAD is 1
[ WARN] [1711424936.619819237]: waiting for image and imu...
... logging to /home/ssz/.ros/log/b94d2e08-eb23-11ee-8e4a-f5e9abdcc60b/roslaunch-ssz-G5-5500-3392020.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ssz-G5-5500:39389/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    rvizvisualisation (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[rvizvisualisation-1]: started with pid [3392081]
`

第二个问题：ekf2为gps和气压计模式时，运行indoor1.launch，使用键盘控制飞机，解锁后，飞机会出现乱飞的情况，请问这是什么原因

l-curious, 2024-05-14:
解决了吗，我vins-fusion 和 三维激光都没rviz显示，二位激光可以

l-curious, 2024-05-17:
是不是ceres和eigen3兼容的问题，

2874160799, 2024-05-19:
> 是不是ceres和eigen3兼容的问题，

是ceres1.14不行还是eigen3不行呢

l-curious, 2024-05-19:
我的是
将如下代码添加到/usr/local/lib/cmake/Ceres/CeresConfig.cmake中：
set(EIGEN_INCLUDE_DIR "/usr/local/include/eigen3")  #添加此行
set(CERES_EIGEN_VERSION 3.3.9)  #添加此行，版本号与你安装的对应

[image]

https://blog.csdn.net/weixin_48093237/article/details/138862043?spm=1001.2014.3001.5501

2874160799, 2024-05-19:
> 我的是 将如下代码添加到/usr/local/lib/cmake/Ceres/CeresConfig.cmake中： set(EIGEN_INCLUDE_DIR "/usr/local/include/eigen3") #添加此行 set(CERES_EIGEN_VERSION 3.3.9) #添加此行，版本号与你安装的对应
> 
> ![a0c8a041ab204da2a7adf21a623c562b](https://private-user-images.githubusercontent.com/145844658/331859260-bc1cf489-8f69-4923-b544-ff81a1da173e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMDk5NzgsIm5iZiI6MTcxNjEwOTY3OCwicGF0aCI6Ii8xNDU4NDQ2NTgvMzMxODU5MjYwLWJjMWNmNDg5LThmNjktNDkyMy1iNTQ0LWZmODFhMWRhMTczZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxOVQwOTA3NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNGI3NzRlNzYxYWUzY2E5ZWQ1NGFjNWM2OTA1Y2Y0ZDcwMDI5ZGRhYTIxOTQyY2I1MzQ1MDE4MGEwM2YwYWQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.PW8b6JTEZ8A9ccpTkzdLfCMYM6awJoWpTQLvUOUk0Q8)
> 
> https://blog.csdn.net/weixin_48093237/article/details/138862043?spm=1001.2014.3001.5501

您好，按照这篇博客的方法，rviz出现了轨迹，但是点云还是没有，不知道是哪里还有点问题
[image]

l-curious, 2024-05-19:
这个我还真没遇到过，要不试试双系统

20246688, 2024-11-25:
请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

20246688, 2024-11-25:
> 解决了吗，我vins-fusion 和 三维激光都没rviz显示，二位激光可以

请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态