robin-shaun/XTDrone #105: 固定翼通信报错WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported

J-starseeker, 2024-04-13:
使用px4 1.13版本，按照教程启动仿真正常，启动通信脚本后仿真程序终端WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported，QGC显示error：SET_POSITION_TARGET_LOCAL_NED coordinate frame 0 unsupported

yuanbovin, 2024-04-27:
在通信python文件 Communication类的初始化中修改以下代码， 加入一行
`        self.target_motion.coordinate_frame = self.coordinate_frame`
[image]

J-starseeker, 2024-04-30:
> 在通信python文件 Communication类的初始化中修改以下代码， 加入一行 ` self.target_motion.coordinate_frame = self.coordinate_frame` ![2024-04-27 18-05-09屏幕截图](https://private-user-images.githubusercontent.com/112998790/326170706-4f8cb1fb-2fc5-4327-b2d3-7f8f62113ba9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0NTc2NDQsIm5iZiI6MTcxNDQ1NzM0NCwicGF0aCI6Ii8xMTI5OTg3OTAvMzI2MTcwNzA2LTRmOGNiMWZiLTJmYzUtNDMyNy1iMmQzLTdmOGY2MjExM2JhOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQzMFQwNjA5MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iOWFmMWVjN2JkYTM3YTBmOTNmYzNmYTYzYTA5MTA1OTI5ZmJkMjEwNDk1MzAxMjMzMTU0NGIyODMyYmZhMDZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.C7BSSBZBvtJSVtbH5CaQz4pp9FN9E6Vvvpir3nrX4gQ)

好的谢谢