robin-shaun/XTDrone #110: make px4_sitl_default gazebo报错

Veast01, 2024-06-06:
错误内容如下：
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/wlxf/PX4_Firmware/Tools/sitl_gazebo/unit_tests

-- Configuring incomplete, errors occurred!
See also "/home/wlxf/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/wlxf/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[234/865] Building CXX object src/lib/...CMakeFiles/bezier.dir/BezierQuad.cpp.o
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/wlxf/PX4_Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/wlxf/PX4_Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/wlxf/PX4_Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[252/865] Building CXX object src/lib/.../avoidance.dir/ObstacleAvoidance.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:230：px4_sitl_default] 错误 1

Veast01, 2024-06-12:
我已解决，卸载Boost重新安装即可。

xiaolutou, 2024-09-03:
我也是这个问题，请问你解决了吗

20246688, 2024-11-25:
请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

20246688, 2024-11-25:
> 我也是这个问题，请问你解决了吗

请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态