robin-shaun/XTDrone #111: roslaunch px4 mavros_posix_sitl.launch无法建立通信

Qserialport, 2024-07-09:
使用的Ubuntu版本为20.04，PX4源码版本为1.14
在使用命令
`roslaunch px4 mavros_posix_sitl.launch`
`rostopic echo /mavros/state`
之后，仍无法建立通信，返回为
```
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
connected: False
armed: False
guided: False
manual_input: False
mode: ''
system_status: 0
---
```
环境变量也设置了，想问一下是不是只支持GAZEBO9以及1.11版本的PX4

DengYaosheng, 2024-07-10:
我记得貌似要改一个portal，，你搜一下，这个是个很常见的问题，请问你试过了吗？

Qserialport, 2024-07-11:
> [DengYaosheng](/DengYaosheng)

我之前设置过端口号，也就是检查mavros_posix_sitl.launch文件内的端口号设置为
`<arg name="fcu_url" default="udp://:14540@localhost:14557"/>`

此外，发现在1.15版本的PX4中，下载的是GAZEBO-CLASSIC11，将.bashrc文件的内容也改正过来了。但是在运行之后仍然连接失败，有警告内容为：
```
libcurl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to fuel.ignitionrobotics.org:443 

libcurl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to fuel.gazebosim.org:443 
```
过了两天之后又跑了一下，发现竟然连接上了，期间没有经过操作

DengYaosheng, 2024-07-13:
走近科学可以拍一年

closed 2024-07-13 by Qserialport

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

Qserialport, 2024-11-25:
> 
尝试一下在QGC中操控无人机进行起飞，倘若在QGC中无人机也出现arm后不起飞，可能需要触地检测参数要修改大一些。

20246688, 2024-11-25:
> > 
> 
> 尝试一下在QGC中操控无人机进行起飞，倘若在QGC中无人机也出现arm后不起飞，可能需要触地检测参数要修改大一些。

嗯嗯呐，已经解决了。好像是ros的依赖项有问题，我重新安装后就正常了。大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

Qserialport, 2024-11-25:
> > > 
> > 
> > 
> > 尝试一下在QGC中操控无人机进行起飞，倘若在QGC中无人机也出现arm后不起飞，可能需要触地检测参数要修改大一些。
> 
> 嗯嗯呐，已经解决了。好像是ros的依赖项有问题，我重新安装后就正常了。大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢

软件在环仿真的时候，GAZEBO里面的无人机轨迹会实时映射到QGC中，通过QGC也是可以实时操控GAZEBO里的无人机的

20246688, 2024-11-25:
> > > > 
> > > 
> > > 
> > > 尝试一下在QGC中操控无人机进行起飞，倘若在QGC中无人机也出现arm后不起飞，可能需要触地检测参数要修改大一些。
> > 
> > 
> > 嗯嗯呐，已经解决了。好像是ros的依赖项有问题，我重新安装后就正常了。大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢
> 
> 软件在环仿真的时候，GAZEBO里面的无人机轨迹会实时映射到QGC中，通过QGC也是可以实时操控GAZEBO里的无人机的

哦哦，原来QGC是这个道理。但是我发现运行配置摄像头的，无人机前面有画面，为啥QGC那个黑色框框木有画面呐大佬

20246688, 2024-11-25:
> > > > 
> > > 
> > > 
> > > 尝试一下在QGC中操控无人机进行起飞，倘若在QGC中无人机也出现arm后不起飞，可能需要触地检测参数要修改大一些。
> > 
> > 
> > 嗯嗯呐，已经解决了。好像是ros的依赖项有问题，我重新安装后就正常了。大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢
> 
> 软件在环仿真的时候，GAZEBO里面的无人机轨迹会实时映射到QGC中，通过QGC也是可以实时操控GAZEBO里的无人机的

还有就是无人机翻到了，卡住了，紧贴建筑模型了，这个怎么处理比较好呢

Qserialport, 2024-11-25:
> 还有就是无人机翻到了，卡住了，紧贴建筑模型了，这个怎么处理比较好呢

进行轨迹规划的时候考虑安全碰撞体积

Qserialport, 2024-11-25:
> > > > > > > 
> > > > > > 
> > > > > > 
> > > > > > 尝试一下在QGC中操控无人机进行起飞，倘若在QGC中无人机也出现arm后不起飞，可能需要触地检测参数要修改大一些。
> > > > > 
> > > > > 
> > > > > 嗯嗯呐，已经解决了。好像是ros的依赖项有问题，我重新安装后就正常了。大佬好，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态。还有就是无人机监控画面怎么投到QGC呢
> > > > 
> > > > 
> > > > 软件在环仿真的时候，GAZEBO里面的无人机轨迹会实时映射到QGC中，通过QGC也是可以实时操控GAZEBO里的无人机的
> > > 
> > > 
> > > 还有就是无人机翻到了，卡住了，紧贴建筑模型了，这个怎么处理比较好呢
> > 
> > 
> > 进行轨迹规划的时候考虑安全碰撞体积
> 
> 好的记本本了，还有就是无人机监控画面怎么投到QGC呢

没有进行相关工作，不了解，QGC只是无人机的地面控制站，可能不可以从QGC里看到图像

20246688, 2024-11-25:
> 没有进行相关工作，不了解，QGC只是无人机的地面控制站，可能不可以从QGC里看到图像

OK

lfh0, 2025-02-26:
> 使用的Ubuntu版本为20.04，PX4源码版本为1.14 在使用命令 之后，仍无法建立通信，返回为`roslaunch px4 mavros_posix_sitl.launch``rostopic echo /mavros/state`
> 
> ```
> header: 
>   seq: 0
>   stamp: 
>     secs: 0
>     nsecs:         0
>   frame_id: ''
> connected: False
> armed: False
> guided: False
> manual_input: False
> mode: ''
> system_status: 0
> ---
> ```
> 
> 环境变量也设置了，想问一下是不是只支持GAZEBO9以及1.11版本的PX4

兄弟，我现在也是这个问题，你最后怎么解决的

Qserialport, 2025-02-26:
> 兄弟，我现在也是这个问题，你最后怎么解决的

并不是，支持GAZEBO11及之后版本的PX4，在.bashrc中设置环境变量后试一下重启虚拟机