robin-shaun/XTDrone #112: 仿真平台基础配置中控制无人机建立通信时报错

DZ-yxz, 2024-07-19:
[ERROR] [1721407031.809526, 2352.013000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication instance at 0x7fefd9af6640>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 117, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 151, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 200, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'
[image]

DZ-yxz, 2024-07-19:
现在是可以连上了，但是地面站显示Amred一会儿之后就变成ready了，也没起飞。速度啥的都设置好了

moyubailan, 2024-08-26:
相同问题

20246688, 2024-10-21:
请问解决了吗

20246688, 2024-10-22:
> 现在是可以连上了，但是地面站显示Amred一会儿之后就变成ready了，也没起飞。速度啥的都设置好了

请问是怎么连接上的？

moyubailan, 2024-10-25:
> 发送日期：2024-10-21 16:47:04
> 抄送人："Fan Kaiwen" ***@***.***>, Comment ***@***.***>
> 主 题：Re: [robin-shaun/XTDrone] 仿真平台基础配置中控制无人机建立通信时报错 (Issue #112)

No,thx

20246688, 2024-10-28:
没有额

20246688, 2024-11-05:
可以打开debug来定位问题，好像是ros文件写入权限没有

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-25:
请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

20246688, 2024-11-25:
> 不，谢谢 发件人：20246688 ***@***.***> 发送日期：2024-10-21 16:47:04 收件人：robin-shaun/XTDrone ***@***.***> 抄送人：“范凯文” ***@***.***>， 评论 ***@***.***> 主题：Re： [robin-shaun/XTDrone] 仿真平台基础配置中控制无人机建立通信时报错 （Issue #112） 请问解决了吗 — 直接回复此电子邮件，在 GitHub 上查看，或取消订阅。 您收到此消息是因为您发表了评论。消息 ID：***@***.***>

请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

20246688, 2025-04-10:
> 解决了嘛

没有呢，请问你有办法吗？