robin-shaun/XTDrone #115: ubuntu 20.04 ros noetic 运行multirotor_communication.py, 再运行multirotor_keyboard_control.py会报错。无法建立通讯

DikaiShen, 2024-09-26:
[ERROR] [1727331187.845439, 2111.633000]: bad callback: <bound method Communication.cmd_vel_flu_callback of <__main__.Communication object at 0x7f6c2a317fd0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "multirotor_communication.py", line 114, in cmd_vel_flu_callback
    self.hover_state_transition(msg.linear.x, msg.linear.y, msg.linear.z, msg.angular.z)
  File "multirotor_communication.py", line 148, in hover_state_transition
    self.hover()
  File "multirotor_communication.py", line 197, in hover
    self.target_motion = self.construct_target(x=self.current_position.x,y=self.current_position.y,z=self.current_position.z,yaw=self.current_yaw)
AttributeError: 'NoneType' object has no attribute 'x'

andy-zhuo-02, 2024-10-13:
这是一个bug，您可以关闭并重启multirotor_communication.py文件，或者先启动键盘控制节点，再启动communication节点

20246688, 2024-10-21:
我也出现这种问题，怎么解决呢

20246688, 2024-10-22:
> 这是一个bug，您可以关闭并重启multirotor_communication.py文件，或者先启动键盘控制节点，再启动communication节点

还是不行

2874160799, 2024-11-04:
> 这是一个bug，您可以关闭并重启multirotor_communication.py文件，或者先启动键盘控制节点，再启动communication节点

我发现当我使用iris_stereo_camera.sdf，不会出现这个错误，当我使用iris_realsense_camera.sdf时，就会报这个错误，场景是indoor1.launch

20246688, 2024-11-05:
> > 这是一个bug，您可以关闭并重启multirotor_communication.py文件，或者先启动键盘控制节点，再启动communication节点
> 
> 我发现当我使用iris_stereo_camera.sdf，不会出现这个错误，当我使用iris_realsense_camera.sdf时，就会报这个错误，场景是indoor1.launch

两者是不是有什么区别呢

DikaiShen, 2024-11-05:
可以先运行通信脚本，再运行位姿真值获取脚本，最后运行multirotor_keyboard_control.py，这样不会报错可以建立通讯

20246688, 2024-11-05:
> 可以先运行通信脚本，再运行位姿真值获取脚本，最后运行multirotor_keyboard_control.py，这样不会报错可以建立通讯

好像是ros文件权限的问题，打开debug可以定位问题，添加权限后就可以键盘控制了

20246688, 2024-11-18:
请问sdf在launch里面配置吗？大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

20246688, 2024-11-18:
请问位子真值脚本在哪里

XXLiu-HNU, 2024-11-18:
same problem

2874160799, 2024-11-18:
> 请问位子真值脚本在哪里
> […](#)

文档里有

20246688, 2024-11-19:
好的，我再试试

  
请问位子真值脚本在哪里
 …
  
文档里有

20246688, 2024-11-19:
还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不

2874160799, 2024-11-19:
> 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不

我没用过，你试试呗

20246688, 2024-11-19:
> > 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不
> 
> 我没用过，你试试呗

在试，快逝了。。。

20246688, 2024-11-19:
> > > > 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不
> > > 
> > > 
> > > 我没用过，你试试呗
> > 
> > 
> > 在试，快逝了。。。
> 
> 两种方式都试过，问题是一样的

怎么搞噢

XXLiu-HNU, 2024-11-19:
> > > > > 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不
> > > > 
> > > > 
> > > > 我没用过，你试试呗
> > > 
> > > 
> > > 在试，快逝了。。。
> > 
> > 两种方式都试过，问题是一样的
> 
> 怎么搞噢

根据你的需求，看是否一定需要使用这个平台，目前我暂时去用px4官网的仿真了

20246688, 2024-11-19:
> > > > > > 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不
> > > > > 
> > > > > 
> > > > > 我没用过，你试试呗
> > > > 
> > > > 
> > > > 在试，快逝了。。。
> > > 
> > > 
> > > 两种方式都试过，问题是一样的
> > 
> > 
> > 怎么搞噢
> 
> 根据你的需求，看是否一定需要使用这个平台，目前我暂时去用px4官网的仿真了

px4官网的仿真是否可以配合qgc软件一起使用呢

2874160799, 2024-11-19:
> > > > > > > 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不
> > > > > > 
> > > > > > 
> > > > > > 我没用过，你试试呗
> > > > > 
> > > > > 
> > > > > 在试，快逝了。。。
> > > > 
> > > > 
> > > > 两种方式都试过，问题是一样的
> > > 
> > > 
> > > 怎么搞噢
> > 
> > 
> > 根据你的需求，看是否一定需要使用这个平台，目前我暂时去用px4官网的仿真了
> 
> px4官网的仿真是否可以配合qgc软件一起使用呢

这个也是在px4基础上的，肯定都是可以用qgc的啊，如果实在不行，就试试用位置真值，再用键盘控制

20246688, 2024-11-19:
好的谢谢啦

20246688, 2024-11-20:
> > > > > > > > 还有就是大佬用没用过文档里面的PX4 1.13版本 一键安装脚本（Beta测试版）？因为我现在是直接安装这个，像之前的环境比如ros什么的就没有重新搭建了。这样合理不
> > > > > > > 
> > > > > > > 
> > > > > > > 我没用过，你试试呗
> > > > > > 
> > > > > > 
> > > > > > 在试，快逝了。。。
> > > > > 
> > > > > 
> > > > > 两种方式都试过，问题是一样的
> > > > 
> > > > 
> > > > 怎么搞噢
> > > 
> > > 
> > > 根据你的需求，看是否一定需要使用这个平台，目前我暂时去用px4官网的仿真了
> > 
> > 
> > px4官网的仿真是否可以配合qgc软件一起使用呢
> 
> 这个也是在px4基础上的，肯定都是可以用qgc的啊，如果实在不行，就试试用位置真值，再用键盘控制

现在好像可以了，好像是EKF参数需要重新清理一下，执行了rm -rf ~/.ros/sitl*就可以键盘控制了。而且也可以飞起来了

20246688, 2024-11-20:
大佬。你知道怎么看四翼无人机运行是GPS定位还是视觉定位或者其他定位呐

20246688, 2024-11-20:
> > 大佬。你知道怎么看四翼无人机运行是GPS定位还是视觉定位或者其他定位呐
> 
> 看EKF那章，或者rostopic list看类似vision这种有没有数据输出，qgc里看ekf参数也行

欧克克

20246688, 2024-11-25:
> > 大佬。你知道怎么看四翼无人机运行是GPS定位还是视觉定位或者其他定位呐
> 
> 看EKF那章，或者rostopic list看类似vision这种有没有数据输出，qgc里看ekf参数也行

大佬桑，请问无人机卡位了有什么好办法解决吗？比如重置如何呢？目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

20246688, 2024-11-25:
请问无人机卡位了有什么好办法解决吗？比如重置如何呢？我目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

andy-zhuo-02, 2025-01-06:
一般来说问题在两个方面，要么是没有连接上MAVROS，要么是没有定位信息。

在1.13.2最新版本的communication文件中，第一个可能性通过提前检测连接状态得到了解决；第二个可能性需要自己去检查mavros/local_position/pose话题，如果没有输出则检查一下是不是之前EKF2融合位姿时改了定位数据来源忘记改回GPS了

andy-zhuo-02, 2025-01-06:
> 请问无人机卡位了有什么好办法解决吗？比如重置如何呢？我目前只看见qgc软件上面有重置模型姿态，但是解决翻面还行，如果是无人机卡死在一个位置上还在转动，重置后无人机还是保持着之前的运行状态

其他问题请单独提ISSUE