robin-shaun/XTDrone #116: 运行三维激光slam outdoor1.launch 时出现模型未加载的情况

2874160799, 2024-10-10:
当我使用 `roslaunch px4 outdoor1.launch` 时，会出现模型未被找到的报错，打开的gazebo地图也有很多模型没被加载出来

[image]
[image]
以下是终端的报错信息

> [Wrn] [ColladaLoader.cc:1441] emission texture not supported
> [Err] [World.cc:2529] Unable to find model[fire_hydrant_95_clone]
> [Err] [World.cc:2529] Unable to find model[fire_hydrant_95_clone_0]
> [Err] [World.cc:2529] Unable to find model[fire_hydrant_95_clone_1]
> [Err] [World.cc:2529] Unable to find model[fire_hydrant_95_clone_2]
> [Err] [World.cc:2541] Unable to find light[user_directional_light_0]
> [Err] [World.cc:2541] Unable to find light[user_point_light_0]
> [INFO] [1728524023.244404, 1752.180000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name iris_0
> [ERROR] [1728524023.245419, 1752.180000]: Spawn service failed. Exiting.
> [ INFO] [1728524023.253111088, 1752.180000000]: Physics dynamic reconfigure ready.
> Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
> Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
> Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
> Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
> [Wrn] [gazebo_gps_plugin.cpp:78] [gazebo_gps_plugin]: iris_0::gps0 using gps topic "gps0"
> [Wrn] [gazebo_gps_plugin.cpp:203] [gazebo_gps_plugin] Using default update rate of 5hz 
> [iris_0/iris_0_spawn-5] process has died [pid 283452, exit code 1, cmd /home/ubuntu/catkin_ws/devel/lib/gazebo_ros/spawn_model -sdf -param model_description -model iris_0 -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 __name:=iris_0_spawn __log:=/home/ubuntu/.ros/log/a79d3d18-86a7-11ef-a6e6-4f0d2e0f45cd/iris_0-iris_0_spawn-5.log].
> log file: /home/ubuntu/.ros/log/a79d3d18-86a7-11ef-a6e6-4f0d2e0f45cd/iris_0-iris_0_spawn-5*.log
> [Err] [InsertModelWidget.cc:403] Missing model.config for model "/opt/ros/noetic/share/turtlebot3_gazebo/models/turtlebot3_autorace"
> [Err] [InsertModelWidget.cc:403] Missing model.config for model "/opt/ros/noetic/share/turtlebot3_gazebo/models/turtlebot3_autorace_2020"
> [Err] [InsertModelWidget.cc:403] Missing model.config for model "/opt/ros/noetic/share/turtlebot3_gazebo/models/turtlebot3_common_meshes"

我去这些路径看了之后好像是模型是有的

[image]
再运行`roslaunch aloam_velodyne aloam_velodyne_HDL_32.launch`无点云画面

andy-zhuo-02, 2024-10-13:
报错提醒的是缺少模型的config文件，检查这些文件夹里面config文件是否存在和正常

2874160799, 2024-10-13:
> 报错提醒的是缺少模型的config文件，检查这些文件夹里面config文件是否存在和正常

这些文件夹的文件我都贴出来了，你可以帮忙看下和正常的是否一样，报错文件夹里没有config文件，但是子文件夹都有的

andy-zhuo-02, 2024-10-14:
正常的结构是模型的文件夹里要包含模型的sdf文件和model.config文件

20246688, 2024-11-18:
大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

2874160799, 2024-11-18:
> 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

i b t

20246688, 2024-11-20:
> > 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢
> 
> 我 b t
大佬。你知道怎么看四翼无人机运行是GPS定位还是视觉定位或者其他定位呐

andy-zhuo-02, 2025-01-06:
> 大佬，我发现我的键盘控制按照文档来还是有问题：我需要打开[qcg]软件，然后再键盘控制按t/y，就有效果。不过无人机arm一会儿后就不转了，需要重新按y再按t才可以转。而且我发现按i调到0.3、0.4，再按b、v这些都不能飞起来。请问是什么问题呢

有其他问题单独提issue，不要在已有issue下提无关的问题

rover11shadow, 2025-09-20:
解决了吗哥